opentx/opentx #1966: telemetry script error

wolkstein, 2015-01-01:
hi,
i get a telemetry script error with an script which run fine on taranis.
i can surly reproduce this error.

1. telemetry script error happened
  1. power taranis and stay on main screen
  2. power the model
  3. switch to telemetry view
  4. display show telem1 script error

2. no telemetry script error
 1.  power taranis and switch to telemetry screen
 2. power the model
 3. telemetry script and and screen works correct.

my telemetry script use an background function. in error case(1) the script background function runs fine and i get audio output, for example my flightmodes. if i switch to the telemetry screen the error happens and also the background function from script is gone.

i try to make an small video with my handy(shaky). however, it demonstrate how i start the taranis and power my model to reproduce the error. also in error case i switch flightmodes before and after i enable the telemetry screen.
the second part of the video show how i power on the taranis and power my model without the 100% reproducible error case.

https://www.youtube.com/watch?v=C6rSBsiqgeE

i try this procedure more than 10 times one after another. and i reproduce the error to 100%.

my script is located here:
https://github.com/wolkstein/MavLink_FrSkySPort/blob/s-c-l-v-rc/Lua_Telemetry/DisplayApmPosition/SCRIPTS/Model_Name/telem1.lua

/g
wolke

projectkk2glider, 2015-01-02:
This might be a problem with memory. Depending on the order of execution the memory usage pattern can be different, so in one case you get out of memory error and sometimes not. There is also issue #1914 that should enable you to see what kind of error is happening.

projectkk2glider, 2015-01-02:
I have now watched your video. The `script syntax error` could also mean `out of memory`. When some kind of error is encountered in Lua script, this script becomes disabled. That is why the background function is also deactivated, because it belong to the same script. Try to reduce script size/memory usage.

projectkk2glider, 2015-01-02:
One memory saving tip for your script: avoid thing like this:
```lua
FlightMode[i].SoundActive1="/SOUNDS/en/AVFM"..(i-1).."A.wav" 
```
and generate strings on the fly in the code when needed.

wolkstein, 2015-01-03:
thx,
i change my script as you suggest. also i remove some not really needed local vars. at least i think that this change free some memory. but the error is there any more:(.
here are my changes.
https://github.com/wolkstein/MavLink_FrSkySPort/commit/7ff6a85c56c5e16f8edf5a12f16a98bfa6f8dc29

also i noticed that taranis statistics give me different "Free Mem" infos. with the old script i read 8192b and after my last changes there are 16384b.
i do not know if this tell me something about available memory for lua scripts. but the difference is around two times more memory available.

it is very strange, because if you power the system that the error must happen, the error happens 100% for sure.
and if you power the system that the error not happens, the script starts 100% for sure without any error case. there is no randomly variance between error or success. 

/g
wolke

projectkk2glider, 2015-01-03:
With that memory amount we can rule out out of memory problem. It must be something in your script, maybe an uninitialized variable.

projectkk2glider, 2015-01-03:
@wolkstein I looked at your script. You have quite a few variables that are uninitialzied. Only one example that could cause your issue:

The variable `prearmheading` is un-initialized, this means its value is `nil` (http://www.lua.org/pil/2.1.html).

That varable is assigned in function:
```lua
	local function armed_status()
	  
	  t2 = getValue(210)
	  apmarmed = t2%0x02
	 
	  --prearmheading =63
	  if apmarmed ~=1 then -- report last heading bevor arming. this can used for display position relative to copter
	    prearmheading=getValue(223)
```
But only if `apmarmed` is at certain value. So if you miss this assignment and try to use it, for example here:
```lua
	local function drawArrow()

	  sinCorr = math.sin(math.rad(getValue(223)-prearmheading))
	  cosCorr = math.cos(math.rad(getValue(223)-prearmheading))
```
You will get an error, example:
```
$ lua
Lua 5.2.3  Copyright (C) 1994-2013 Lua.org, PUC-Rio
> local foo
> print (5-foo)
stdin:1: attempt to perform arithmetic on global 'foo' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: in ?
```

bsongis, 2015-01-03:
Also if the GPS doesn't have a fix, I am not sure what will do this line:
pilotlat = math.rad(getValue("latitude"))

projectkk2glider, 2015-01-03:
Correct, without GPS fix the `getValue("latitude")` will return **nil** - ie no return value.

bsongis, 2015-01-03:
... and rad(nil) could raise an error ...

wolkstein, 2015-01-04:
thx for pointing out,
if i understand this right the script can throw errors because for example arithmetic operations with not initialised(nil) variables. this mean if no receiver sends telemetry to the taranis for example the model is not powered the script must crash because some of the vars are at this time NIL?

wolkstein, 2015-01-04:
ok,
i try it now with initialized variables. 
changeset:
https://github.com/wolkstein/MavLink_FrSkySPort/commit/64b0c10402aa53ca6619eccd17cecacbd619b9f9

after this changes the error is gone :).
currently without extra checking against NIL return values from getValue(xy).

for checking against NIL my idea was a simple check function for numeric valus.
my idea
```lua

-- for example check getValue(223) 
prearmheading=checkNILNum(getValue(223))

local function checkNILNum(x)
  if (x!=nil) then 
    return x
  else
   return 0 
  end
```
could such a check prevent the NIL problem for arithmetic operations?
but after the last changeset the script also work fine without such a check.
 
thx a lot for helping me.

closed 2015-01-04 by wolkstein

projectkk2glider, 2015-01-04:
The not equal operator is `~=` not `!=`, otherwise your function is good

wolkstein, 2015-01-04:
:) yes, i have correct this in my last comment before you comment the syntax error. only for followers. i firstly write "if (x!=nil) then".