opentx/opentx #1969: Companion Simulator.exe not starting

projectkk2glider, 2015-01-03:
Having just tried latest nightly in Windows XP the Companion's standalone simulator won't start. I get:
```
C:\Program Files\OpenTX>simulator.exe
The system cannot execute the specified program.
```
The exe is there and looks like a normal win32 program. I remember having this issue a while back, but did not make a fuss then.

My self build Companion simulator on Linux works without any problems.

projectkk2glider, 2015-01-03:
I used http://dependencywalker.com/ and found out that I am missing: IESHIMS.DLL and WER.DLL

projectkk2glider, 2015-01-03:
The missing dll's are not the problem! Companion.exe has same missign dll's and its working. Also installing Microsoft Visual C++ 2010 Redistributable Package (x86)  did not help.

kilrah, 2015-01-03:
http://jenkins.open-tx.org/nightly-20/01_03_2015/companionInstall_2.0.14.exe works fine here, that's on W7 though.

projectkk2glider, 2015-01-03:
Yes it seems to be WinXP isssue.

projectkk2glider, 2015-01-03:
Interesting message from Windows event log:
```
Dependent Assembly Microsoft.VC90.CRT could not be found and Last Error was The referenced assembly is not installed on your system.
```

projectkk2glider, 2015-01-03:
Success, I had to install Microsoft Visual C++ 2008 Redistributable Package (x86) from http://www.microsoft.com/en-us/download/confirmation.aspx?id=29

closed 2015-01-03 by projectkk2glider