opentx/opentx #1970: Build fails with PCB=9X2561 and PWM_BACKLIGHT=YES

kh90909, 2015-01-03:
It looks like the PWM_BACKLIGHT code didn't get updated with ATmega2561-specific defines and code, so there are compile errors related to the different timer registers.

Happy to fix this myself and submit a pull request, but I could use some help to determine which timer I should use. The current mega64/128 code uses timer 2. The [2561 migration app note] (http://www.atmel.com/Images/doc2581.pdf) suggests that its timer0 is equivalent to timer2 on the 64/128, but I'm concerned that timer0 might be used elsewhere. Someone familiar with the code will be able to answer this much faster and more reliably than me.

bsongis, 2015-01-04:
I would say Timer4!

bsongis, 2015-01-04:
Yes you have to use Timer0 which is unused (on 9X2561) if I am not mistaken

bsongis, 2015-01-04:
When confirmed it will need to be ported to 'next'

kh90909, 2015-01-04:
Thank you very much for taking the time to fix this so quickly. I'll test it and report back.

kh90909, 2015-01-08:
Sorry, it took a while to find the time to get this done.

The original quick fix had a typo in the CS2:0 value and didn't account for the SP22 flag for the inverted backlight with the Smartieparts v2.2 programmer board. The above pull request fixes these issues.

I've tested this on my 9X2561 transmitter and it controls the backlight brightness correctly, but I won't be able to do a flight test for some time. However, I have reviewed the code and confirmed that timer0 is not used elsewhere on 2561 boards (except for the audio/voice modules, which are incompatible with PWM_BACKLIGHT), so the chances of any unintended side effects are very minimal.

bsongis, 2015-01-08:
Thanks for your tests, and sorry for the typo ;)

bsongis, 2015-01-08:
Would you open a pull request with this fix ported to 'next' (next release 2.1.0)? Thanks!

kh90909, 2015-01-08:
Sure, I'll do that this evening.

PR #1984 (2015-01-08): Tested fix for PWM_BACKLIGHT on ATmega2561 MCUs
This fixes #1970.

The functionality of timer0 and timer2 are switched between ATmega64/128
and 641/1281/2561. When compiled for ATmega2561, this commit sets up timer0
in the same way as timer2 on ATmega64/128 (phase correct PWM, clk/64 count
rate, inverted if SP22 is defined).

The previous quick fix had a typo in the initialization of TCCR0B (0b111 is
an invalid value for the clock select bits CS2:0). It also did not respect
the SP22 define which indicates the PWM must be inverted. Smartieparts
v2.2 or later programmer boards have an inverted b [... cut]
PR #1988 (2015-01-09): Port ATmega2561 PWM_BACKLIGHT fix from master branch
Fixes #1970 for the next branch.
closed 2015-01-09 by bsongis