opentx/opentx #1973: T+ lcd colour change on flashing v2.0.13

JonnyG30, 2015-01-05:
T+ started with dark blue LCD backlight, after flashing v2.0.13 the backlight switched to dark orange???. can't find a way to switch back. "backlight colour" setting does not change it back (produces washed out light blue (almost white)). 

Is support available for full colour spectrum of backlight?? if not can it be??

thanks

kilrah, 2015-01-06:
It's different hardware. There's a version with blue and white, and one with amber and white. Color selection works between the 2.

closed 2015-01-06 by kilrah

JonnyG30, 2015-01-06:
I thought the same thing and was very surprised when it happened BUT the fact of the matter is my taranis+ had a dark blue screen, I played with it for an hour before flashing so i know it was blue!!!. Then after I finished toying around with it I flashed with the firmware mentioned above and it's now dark orange!!!! 
It can't be hardware fixed or my issue would be impossible. if there is a way to prove my issue tell me what to do and i will gladly do it.