opentx/opentx #1975: Taranis S-Port A3 (and A4?) incorrect scaling?

MikePRoberts, 2015-01-06:
I am getting a strange scaling when I changed from sending analogue data as A3 rather than A2 (to avoid clash with A2 on the X4R receiver).  Using a range of 25.5v on the Taranis I get a starting display value of 7.7v from zero in, increasing to 27.4v (that displays as 1.8v with the overflow that occurs) from 0xFF in.  S-Port (high byte)/Taranis Display examples are: 0/7.7; 48/11.4; 100/15.4; 149/19.2; 198/23.0;  230/25.5; 255/1.8 (or 27.4 adding in the overflow).  Note there may be a final digit error as I took the pairs of readings a few seconds apart.  Checked with Taranis with Firmware 2.0.13 both with X4R and X8R receivers.  Please let me know if I can do further tests to help.

MikePRoberts, 2015-01-09:
Damjan

I checked with 2.0.12 and 2.0.13.  I also saw processing of (packet*256 +165)/330 in the current master code.  However the data converts closer to (packet*256 +99)/330.  Please recognise I am not all all familiar with the details of the processing.  I have however a microcontroller that connects to S-Port and I can quickly program it to send specific data if that helps.

Mike

From: Damjan Adamic 
Sent: Friday, January 09, 2015 11:35 AM
To: opentx/opentx 
Cc: Mike Roberts 
Subject: Re: [opentx] Taranis S-Port A3 (and A4?) incorrect scaling? (#1975)

@MikePRoberts which version of opentx does this?

kilrah, 2015-01-09:
What are you actually using to send the A3 and A4 values to the receiver? I don't see anything that mention what sensor you actually use.

MikePRoberts, 2015-01-09:
Andre

This is my own vario.  I send voltage as well as vspd and alt as in most cases I want to monitor the battery as well.  I originally sent the voltage data as A2 but thought I would switch to using A3 as the new X4R Rx has its own A2 input and I thought I should avoid a clash.
In case you are interested I am using the MS5611 sensor that is amazing (as are the calculations required).  On a calm day it will give your height within 0.1m!  This is the same sensor as the OpenXvario project but I am using a PIC (using assembler) and have included a very special filtering algorithm that almost doubles the response speed with the same smoothing.  It is also just 11x 30mm size.
I have done a fair bit with both the D series and the S-Port telemetry.
Please let me know If I can help you at any time.  Note I am not any good at C.  I program almost entirely in assembler.
Mike
From: Andre Bernet 
Sent: Friday, January 09, 2015 2:46 PM
To: opentx/opentx 
Cc: Mike Roberts 
Subject: Re: [opentx] Taranis S-Port A3 (and A4?) incorrect scaling? (#1975)

What are you actually using to send the A3 and A4 values to the receiver? I don't see anything that mention what sensor you actually use.

kilrah, 2015-01-09:
>I originally sent the voltage data as A2 but thought I would switch to using A3 as the new X4R Rx has its own A2 input and I thought I should avoid a clash.

OK. Then I suspect you just changed the data ID from the one for A2 to the one for A3? If yes then it's normal for it to be incorrect. A3/A4 do not use the same scaling as A1/A2, so you need to change your code to match. See OpenTX code for ratios.

projectkk2glider, 2015-01-09:
Let me help you:
```c
        else if (appId >= A3_FIRST_ID && appId <= A3_LAST_ID) {
          frskyData.analog[TELEM_ANA_A3].set((SPORT_DATA_U32(packet)*255+165)/330, UNIT_VOLTS);
        }
        else if (appId >= A4_FIRST_ID && appId <= A4_LAST_ID) {
          frskyData.analog[TELEM_ANA_A4].set((SPORT_DATA_U32(packet)*255+165)/330, UNIT_VOLTS);
        }
```

Basically A3 and A4 range if from 0 to 330 which is internally scaled by opentx to 0-255.

MikePRoberts, 2015-01-09:
Andre

Yes I did just change the data ID. Let me do some more tests.

Why change to a range of 330?  The original measurement of voltage will give a number with range of 255 for 8 bit or 511, 1023 for 9,10 bit.  So in order to get a range of 330 one has to either use part of the ADC range losing resolution or multiply up to 330 adding complication and rounding errors when the final number will be scaled anyway at the transmitter. Surely a range of 2^n-1 is better.

Thank you

Mike

From: Andre Bernet 
Sent: Friday, January 09, 2015 6:34 PM
To: opentx/opentx 
Cc: Mike Roberts 
Subject: Re: [opentx] Taranis S-Port A3 (and A4?) incorrect scaling? (#1975)

  I originally sent the voltage data as A2 but thought I would switch to using A3 as the new X4R Rx has its own A2 input and I thought I should avoid a clash.

OK. Then I suspect you just changed the data ID from the one for A2 to the one for A3? If yes then it's normal for it to be incorrect. A3/A4 do not use the same scaling as A1/A2, so you need to change your code to match. See OpenTX code for ratios.

kilrah, 2015-01-10:
A1/A2 are a single byte, i.e. 0-255 where 255 is 3.3V at the input.
For A3/A4 as the whole smart port protocol is available FrSky decided to describe it as V*100 (i.e. 3.3V = 330), which makes sense as the resolution is increased and it's clearer.

MikePRoberts, 2015-01-10:
Andre
Please reconsider.  This seems a little crazy.  No user sees the 0-330 values – only the programmers.  The end value for processing in opentx is a number from 0-255.  The starting number will also be a number from 0-255 (using the first 8 bits from 8,9 or 10 bit ADC – there being no point processing more bits than are used).  So in order to have the nicety of the transmitted number having increments of 0.01v we have to process the original number with: a 2 byte by 1 byte multiplication, a 1 byte to 3 byte addition and (worst) a 1 byte into 3 byte division.  And then only to do the reverse when the data arrives.  In any case no-one will want to measure a max of 3v3, so an input voltage divider will always be used.  From the end users perspective there is no benefit.  So why add this complication?  For both the current system for A2 and the new for A3, A4 one simply has to calculate resistors to give 3v3 at the maximum desired range and then set the range in the transmitter.  This is adding complication at both the sensor and opentx than no user sees or gains any benefit from.  It is not clearer and the resolution is not increased (ends up as single byte).
This looks like it has not been thought through.
If one wants to increase resolution then go to 9 or 10 bit resolution.  This is simple and adds a little value (I find 0.1v resolution over 0-25.5v more than adequate).
If one wants to make it clearer then maybe remove the conversion back to 0-255 and for A3, A4 have the end user enter the voltage divider ratio rather than voltage range.  But this still does not seem much help as the end user still has to multiply the divider ratio by 3v3 to check he has adequate voltage range.  The current system of sensor suppliers specifying the voltage range and users entering this value into opentx seems the clearest option.
Is my argument sensible?
Thank you for your outstanding work.
Mike
From: Andre Bernet 
Sent: Saturday, January 10, 2015 5:52 AM
To: opentx/opentx 
Cc: Mike Roberts 
Subject: Re: [opentx] Taranis S-Port A3 (and A4?) incorrect scaling? (#1975)

A1/A2 are a single byte, i.e. 0-255 where 255 is 3.3V at the input.
For A3/A4 as the whole smart port protocol is available FrSky decided to describe it as V*100 (i.e. 3.3V = 330), which makes sense as the resolution is increased and it's clearer.

MikePRoberts, 2015-01-10:
My suggestion here would be to left justify the ADC result so in all cases a 3v3 input gives 2^32.  This is future-proof and can easily take different ADC resolutions without changes to the transmitter settings.  So if there is an 8 bit ADC then the result sits in the high byte of the 4 byte data.  If a 10 bit ADC is used the value sits in the top byte plus the top 2 bits of the second byte.  No change in range setting is required but the extra bits deliver additional resolution.  This also enables consistency with current A1 and A2 in that the end user sets the Range value in opentx.
Mike

kilrah, 2015-01-10:
There is nothing to reconsider. FrSky have decided to go with V*100 for their RPM/voltage sensor that's been released months ago, that's just how it is, nothing can be changed anymore.

What needs improvement is the handling on OpenTX side to avoid scaling, but that's another issue and will be part of 2.1. We learned about the change in scaling literally days before release of 2.0... so we had 2 choices, either extend the current Ax code which was already scalable to add 2 more items and do the scaling as a workaround, or recode everything. The latter would have been way too long for the available time, and the work to do it cleanly would have had to be scrapped shortly after as telemetry is being completely reworked in 2.1 anyway.

closed 2015-01-10 by kilrah

bsongis, 2015-01-10:
And is it ok now in 2.1? This needs to be tested!