opentx/opentx #1979: T+ no sd card mount, no companion eeprom read/write in bootloader mode

JonnyG30, 2015-01-07:
running windows 8.1. using openTX V 2.0.13

windows gives confirmation taranis+ is connected (connection sound and USB eject icon), driver shows up in device manager (installed using zadig). I get no SD card mount and no eeprom read/write ability in companion when taranis+ is in bootloader mode. 

when taranis+ is connected whilst OFF, firmware can be flashed without issue. 

I've tried rolling drivers back, deleting and re-installing drivers, a couple of re-flash's, connecting powered on, connecting powered off, connecting in bootloader mode and scouring the internet for fixes with no fix found that isn't something i've already tried. I'm out of things to try :(

kilrah, 2015-01-07:
You probably used Zadig to install the driver on the wrong device. Turn the radio on in bootloader mode, connect to USB, and remove the device (with "Remove driver" checked) in the device manager. Disconnect and reconnect and Windows should reinstall the mass storage driver.

In any case that's not a software issue.

closed 2015-01-07 by kilrah

JonnyG30, 2015-01-07:
already tried that and it didn't work. 

the issue persists using microsoft automatic driver selection.

donjakobo, 2015-10-09:
I followed @kilrah 's advice, and removed the usb driver titled "FrSky Taranis Mass Storage" checked "remove driver" and then it mounted correctly. Able to get back in via Companion 2.1.3