opentx/opentx #1982: Channel monitor limited to +/-125 instead of +/-150 with extended throws

pindeslandes, 2015-01-07:
Taranis – Open Tx 2.0.13
The title says all but a picture may help:
[image]

franck76, 2015-01-07:
with Taranis - Open TX 2.0.13
Channel monitor can display [-150,+150]
but you have to set up servo menu with Min = -150, Max = +150.
[image]

kilrah, 2015-01-08:
Yeah but the bar reaches the end at 125. As far as I remember it was decided to leave it like that.

projectkk2glider, 2015-01-12:
We need decision, do we change graph to 150 or leave it as it is? The change (I presume) is minimal and very simple.

bsongis, 2015-01-12:
Nobody uses 150% is it needed?

ckleanth, 2015-01-13:
If you got servos with no endstops (dunno robotics or animatronic applications) then it would be useful. Personally I don't/cant use more than 125%

kilrah, 2015-01-13:
On servos with no endstops it makes even less sense, you never need more than 20% or so, anything beyond that is "full speed".

bsongis, 2015-01-13:
So what do we decide on this one? For the moment I vote for no change

ArchmageAU, 2015-01-14:
What makes sense to me is that since the graph changes between extended and normal limits, the graph for extended limits should max at the limits (+/-150) instead of another arbitrary historical figure (+/-125).

kilrah, 2015-01-14:
But then everybody loses some resolution on the graph when extended limits are on for the 2-3 people who might use the 125-150 range...
I vote for no change too. It can indeed be seen as a discrepancy, but I see it as a good compromise for general use.

InRVoice, 2015-01-14:
graph resolution doesn't matter. % value is on the left. I use more than 125% and I think it is a benefit to have this degrees of freedom instead of limitations like comercial systems.

pindeslandes, 2015-01-15:
Agreed! And it would be logical and consistent, as expected.

franck76, 2015-02-03:
Why not add max value (125 or 150) as an option to extended limits
and allow the bars to reach the end at this maximum?

closed 2015-03-30 by kilrah