opentx/opentx #1985: Improve operation of lost antenna warning

mpjf01, 2015-01-08:
Replacement of the standard Taranis antenna with a screw on version (coax either soldered to the back board or fitted to a UFL connector on the back board) with a SMA connector to enable easy antenna replacement with higher db versions is a common enhancement supported by vendors.

There is a warning issued if the firmware detects that no antenna is connected to the RF module.This warning will appear if there is no coax connected to the back board, but does not occur if there is coax attached, regardless of whether or not the SMA connector or aerial is connected. If it is possible to detect the lack of a screw on aerial, or the failure of the connector or coax, then this should be incorporated in the interests of safety.

kilrah, 2015-01-08:
Nothing can really be done about it. The module measures reflected power that comes from the antenna port, and if there's too much it triggers the warning. Your open SMA probably already radiates enough for the warning not to appear. If the warning level was lowered, then there will be false warnings on normal antenna setups. The detection circuit is not very precise and not intented to detect more than "nothing connected" or "broken coax".

closed 2015-01-08 by kilrah

mpjf01, 2015-01-08:
It is unfortunate that nothing will be done about it. If the warning is intended to detect "nothing connected" or "broken coax" as you say, then it fails - unless the breakage is at the point where the coax is connected to the backboard and both conductors disconnect.

projectkk2glider, 2015-01-09:
The SWR value comes from internal XJT module that has proprietary FrSky firmware on it. There is nothing OpentTX code can do to improve that measurement. You should talk to FrSky about this issue.