opentx/opentx #1986: Poor VFAS voltage resolution from Cleanflight

maca55, 2015-01-08:
I'm using Cleanflight to send voltage back to Taranis from a Naze32 and am disappointed with the accuracy of the VFAS readings.

The resolution on the VFAS reading displayed on Taranis appears to be around 0.2V. This results in voltage display error of as much as 0.2V. With some minor changes to Cleanflight and openTX we could reduce this error to +/-0.05V. Or even better if we could send the value with 0.01V resolution and reduce error to under 0.01V.

The reason for the poor resolution and accuracy appears to be a combination of lack of rounding of values in Cleanflight and also the use of the scaling factor used for VFAS value on both ends "21/110". 

I notice that openTX already has some support for more accurate Voltage readings using "FAS_BSS":

#if defined(FAS_BSS)
      frskyData.hub.vfas = (frskyData.hub.volts_bp * 10 + frskyData.hub.volts_ap);
#else
      frskyData.hub.vfas = ((frskyData.hub.volts_bp * 100 + frskyData.hub.volts_ap * 10) * 21) / 110;
#endif

Would it be possible to have Cleanflight VFAS values treated differently, ie) as the more accurate kind? Perhaps similar to way openXSensor used the 0x39 ID for transmitting more accurate readings?

We could get rid of the 21/100 scaling and ideally move to sending and receiving the voltage with 0.01V resolution, the same as with the individual cell values.

projectkk2glider, 2015-01-09:
First thing you can do is to send VFAS as id 0x39, this is used for example in Openxsensor project and supports values with 100mV resolution. There is no other possibility with FrSky D type telemetry (which I presume is used by Cleanflight).

However if you switch to SPORT telemetry format, then protocol itself supports 10mV precision for VFAS, but OpenTX scales this down to 100mV resolution. But this can be fixed ,there is already todo for this in the code.

bsongis, 2015-01-09:
FAS_BSS is used for one VFAS I have here with a different resistor divider, such sensors were not commercialized by FrSky.

And yes CleanFlight should implement S.PORT, it's perhaps not so simple !

closed 2015-01-09 by bsongis