opentx/opentx #1987: lcd.drawLine and lcd.drawPixmap issue

lakeroe, 2015-01-08:
Using following small LUA script (in OpenTX 2.0.13)

lcd.drawPixmap (0, 0, "/SCRIPTS/BMP/quad.bmp")	-- Size is 20x20
lcd.drawPoint  (0, 0)
lcd.drawPoint  (19, 0)
lcd.drawPoint  (0, 19)
lcd.drawPoint  (19, 19)
lcd.drawLine   (0, 21, 19, 21, SOLID, FORCE)

gives this output
[image]

In my optinion this should be right
[image]

Am I wrong or can anybody reproduce this issue ?
I think lcd.drawLine doesn't draw the first pixel and lcd.drawPixmap doesn't draw the last colum.

projectkk2glider, 2015-01-09:
We need your bitmap file to test this.

projectkk2glider, 2015-01-09:
`lcd.drawLine()` indeed has a bug, first pixel is not drawn. However  I can not reproduce a bug with `lcd.drawPixmap()` with test image https://raw.githubusercontent.com/opentx/opentx/master/radio/src/tests/4b_31x31.bmp

lakeroe, 2015-01-09:
I also can't reproduce the issue using your test image.
My test image is here:
http://www.adrive.com/public/EsNJgQ/quad.bmp

projectkk2glider, 2015-01-09:
Whit your image I also reproduced this bug. Will take care of this one to...

projectkk2glider, 2015-01-09:
All done

closed 2015-01-09 by projectkk2glider

projectkk2glider, 2015-01-09:
Ported to next in b878dc673674c9e4669f77f1f927e14d3a240f78

commit 6d46d52b (2015-01-09): Fixes #1987: bmpLoad() last column not loaded for some bitmap widths, gtests added
commit a18a79bc (2015-01-09): Fixes #1987: missing bitmap for gtests
PR #1989 (2015-01-09): Projectkk2glider/issue 1987
As a bonus, second commit saves 56 bytes of FLASH.

Closes #1987
commit 10b1836a (2015-01-09): Fixes #1987: gtests for lcd_line()