opentx/opentx #1990: Enhancement: Set Model timer value from custom functions

Hwurzburg, 2015-01-10:
While you can directly set a timers value, you cannot set the model default...in my case, I sense a switch position on power up for different battery or prop configurations and change curves, etc....also set the flight timer(s)....this works well at POR....but using the "reset flight" feature as I put the next battery in the quad to fly, will reset the timers to the model page's default....it would be nice to change that value for a given power up session, so that I could just reset the flight for a new battery, rather than cycle the power switch...

kilrah, 2015-01-10:
Already there.

[image]

closed 2015-01-10 by kilrah

Hwurzburg, 2015-01-10:
You did not understand the request...please read the entire post...
using "Set TImer x" does so only temporarily and is exactly what the first sentence was referring to...." Reset Flight" resets the timers to the default value on the Model setup page....the enhnacement would be to set THAT value from a custom function...so "reset Flights" use it again without having to toggle SA, in your example....

projectkk2glider, 2015-01-10:
Then use Lua script to set timer to whatever configuration you want.

kilrah, 2015-01-10:
Or use a switch to trigger both a Reset flight SF, then a Set Timer one when it's released for example.

Hwurzburg, 2015-01-10:
good idea! thnx...saves having to learn another programming language...