opentx/opentx #1995: [Taranis] [enhancement] Limitation of Taranis buzz/hum

LapinFou, 2015-01-12:
Hi folks,

I helped a friend who owns a Taranis B. As many Taranis, there is a constant buzz (which is not completely fix, but much much better on my Taranis+). Last night I thought about a simple software solution to limit this problem. I searched on the closed issues, and didn't find anything. So, sorry if it has already been discussed.
Here is the proposal, could it be possible to completely mute the speaker when no sounds need to be played? Is it possible that the firmware enables the audio ONLY when a sound needs to be played?
If this mechanism is not already implemented, it (maybe) could improve/limit this annoying buzz/hum?
Off course, I don't know if this could have an electronic impact on the audio amplifier (components stress for example).

Thanks!

By the way, happy new year to the whole team. :)

projectkk2glider, 2015-01-12:
The RF noise is coupling onto the output stage of audio amplifier. Therefore whatever we do on the input side wont change the noise. And we can't control audio amplifier form CPU in any way, it is always turned on.

LapinFou, 2015-01-12:
OK. Thanks for the clarification. I thought that the "volume function" was controlling the audio amplifier output stage. Since it is not the case, then this enhancement is not possible.

bsongis, 2015-01-12:
No, not possible !!!

closed 2015-01-12 by bsongis