opentx/opentx #1997: Problems loading firmware with Companin 2.0.14

jtaylor2, 2015-01-13:
Same problems exist with downloaded Companion 2.0.14 on Windows 7 and self compiled Companion from tag 2.0.14 on openSuSe 13.2. to a Rev. B Taranis
  
1.  Firmware load with Taranis mounted as removable device completes, but Taranis hangs and won't power off.  Have to pull battery to power off.  Turning on and hangs at splash screen and won't power off.  Tried loading opentx-taranis-lua-nooverridech-en-2.0.14.bin and opentx-taranis-lua-nooverridech-en-2.0.13.bin and both hang and won't power off.

2.  Copy opentx-taranis-lua-nooverridech-en-2.0.14 as opentx-2_0_14 to FIRMWARES on SD and update completes and Taranis boots but has default splash screen instead of custom.

3.  Flash firmware from 2.0.14 Companion using DFU and Taranis works, but again have default splash screen instead of custom.

4.  Flashed opentx-taranis-lua-nooverridech-en-2.0.14.bin firmware using Companin 2.0.13 with Taranis mounted as removable device and flash completed, Taranis worked, and I had my custom splash.

Jim

projectkk2glider, 2015-01-13:
Confirmed, the firmware updating in Companion 2.0.14 in **Bootloader mode does nothing**. The DFU mode (radio off) on the other hand works for firmware update.

 Upload and download of EEPROM works in Bootloader mode.

bsongis, 2015-01-13:
On Linux as well? The only thing I changed is the open in READ/WRITE mode. Does it work with a simple cp (copy on Windows)?

bsongis, 2015-01-13:
I did this commit:
https://github.com/opentx/opentx/commit/e74ab6f15bf85805cb825eba9f14775c7c10dc38

But given that problems happen in bootloader from 2.0.13 I guess it's only the first change (ReadWrite instead of WriteOnly) which is wrong, I don't know why

projectkk2glider, 2015-01-13:
I am testing on Linux. Two things broken:
* no matter what mode of update, if one chooses a library (i did not test custom) splash screen, nothing is done. In DFU mode the update actually writes to the radio, but i see no change in firmware even if I write 2.0.13. Looks like it always writes back readed version with default splash. 
* using default splash screen ony DFU mode works.
* if `check firmware compatibility` is checked, then firmware file (in /tmp dir) is 500kB, if unchecked the firmware size is about 300kB.

Bottom line it works in all cases if  `check firmware compatibility` is **unchecked**.

bsongis, 2015-01-13:
Then this commit? 
https://github.com/opentx/opentx/commit/026b0f141ed1d90b8d08cfe3acad3daf9423ee4f

projectkk2glider, 2015-01-13:
>I hope that the unique filename is really unique!

It is not, only the current PID is appended to the given file name. 
```c
QString generateProcessUniqueTempFileName(const QString & fileName)
{
  QString sanitizedFileName = fileName;
  sanitizedFileName.remove('/');
  return QDir::tempPath() + QString("/%1-").arg(QCoreApplication::applicationPid()) + sanitizedFileName;
}
```

I can fix this, but I am not near the radio until this afternoon. We just need to use another file name for compatibility check.

 I posted short explanation on rcgroups http://www.rcgroups.com/forums/showthread.php?t=2178865&page=455#post30465986

projectkk2glider, 2015-01-13:
Testers needed, the Windows Companion containing this fix is here: http://jenkins.open-tx.org/nightly-20/01_13_2015/companionInstall_2.0.14.exe

jtaylor2, 2015-01-13:
Works for me.  Windows 7 64 Bit Home Premium.

Jim

bsongis, 2015-01-13:
Fixed, right?

bsongis, 2015-01-13:
Still needs to be ported to 'next' ant it will be marked as fixed.

jtaylor2, 2015-01-14:
I still have a problem flashing with Companion 2.0.15 on my Linux system.  Companion 2.0.15 on my Windows 7 system works fine.

Companion 2.0.15 self compiled on openSUSE 13.2 from git checkout tag 2.0.15.  Loading downloaded (not self compiled) opentx-taranis-lua-nooverridech-en-2.0.15.bin to a rev. B Taranis.

If I flash to the Taranis mounted as a removable device the flash completes, umount the device's, hit exit and the Taranis hangs and won't power off.  Pull battery, power on, backlight comes on, no splash, no sound and hang and won't power off.  This is the same problem I had with 2.0.15 flashing this way.  Hangs the same regardless if check for compatibility is checked or unchecked.

Flash with Companion using DFU to powered off Taranis works fine.  Flash by copying firmware to FIRMWARES and flashing using the bootmanager works fine.

Perhaps it's something with my build environment, but I have been self compiling every version since r2923 and 2.0.14 was the first one where I had a problem.

Jim

projectkk2glider, 2015-01-14:
The bootloader  had a bug where it would hang just like you describe. Are you doing 'make clean' before the make. Only then the Bootloader gets rebuilt.

Now that your flashed official 2.0.15 ib dfu mode, you should not have this problem anymore. Can you test, by re-flashing 2.0.15 with some other splash screen?

projectkk2glider, 2015-01-14:
Users on rcgroups report success with opentx 2.0.15, which includes this fix. I will port this to `next` and close it.

jtaylor2, 2015-01-14:
Yes I did a make clean before building.  Tried again with 2.0.15 bootloader with same results (hang).  Also tried with default splash and same hang.  Tried flashing 2.0.12 with Companion 2.0.15 and Taranis wouldn't even power on after flash, had to flash with DFU to recover.

Not a big problem, since DFU and copy to SD methods work, as well as flashing from Windows, but curious since it worked up through 2.0.13 and problem only appeared with 2.0.14.

Not worth keeping issue open if I'm the only one having the problem, but I'm curious and will pursue it when I get some time.

Jim

projectkk2glider, 2015-01-14:
I will complile Debian pacakes later today and put them on the server. It would be good if you then use that package, install it and test again. Maybe it is something in your build. 

What can you do is to make a debug version of Companion (`cmake  -D CMAKE_BUILD_TYPE=Debug ../src`) and then run it from the command line (`./companion`). When you do the firmware update you will get some debug prints on console, I would like to see them here.

jtaylor2, 2015-01-14:
Not sure SUSE can install a Debian package, but I'll check.  Output from debug build pasted below.  I don't see anything out of the ordinary.  And even more confusing I copied the firmware from the Taranis before and after and they are the same (md5sums match) and still it hangs and I have to pull the battery.  See pasted below, FIRMWARE is before and FIRMWARE1 is what was flashed and hangs.

ThinkCentre:~/frsky/opentx/git/opentx/companion/lbuild # ./companion 2>&1 | tee companion.out
MainWindow::writeFlash(): started with "" 
burnDialog::on_BurnFlashButton_clicked(): patching "/root/frsky/taranis/firmware/opentx-taranis-lua-nooverridech-en-2.0.15.bin" with custom splash screen and saving to "/tmp/3626-flash.bin" 
MainWindow::writeFlash(): flashing  "/tmp/3626-flash.bin" 
MainWindow::writeFirmwareToRadio(): writing "/mnt/taranis/eeprom/FIRMWARE.BIN" from "/tmp/3626-flash.bin" 
MainWindow::writeFlash(): removing temporary file "/tmp/3626-flash.bin" 

ThinkCentre:~/frsky/companion/work # ls -la
total 1104
drwxr-xr-x  2 root root   4096 Jan 14 10:51 .
drwxr-xr-x 20 root root   4096 Jan 14 10:34 ..
-rwxr-xr-x  1 root root  32768 Jan 14 10:35 EEPROM.BIN
-rwxr-xr-x  1 root root  32768 Jan 14 10:51 EEPROM1.BIN
-rwxr-xr-x  1 root root 524288 Jan 14 10:35 FIRMWARE.BIN
-rwxr-xr-x  1 root root 524288 Jan 14 10:50 FIRMWARE1.BIN
ThinkCentre:~/frsky/companion/work # md5sum FIRMWARE.BIN
a7d44104e57a0306f0f7341c933f8053  FIRMWARE.BIN
ThinkCentre:~/frsky/companion/work # md5sum FIRMWARE1.BIN
a7d44104e57a0306f0f7341c933f8053  FIRMWARE1.BIN

projectkk2glider, 2015-01-14:
Why do you have such a weird path to Taranis firmware file `/mnt/taranis/eeprom/FIRMWARE.BIN`? Can you show the output of `mount` when radio is connected in Bootlader mode.

bsongis, 2015-01-14:
I try to guess ;) He has an eeprom directory on his SD (labeled taranis) and in this folder there is a FIRMWARE.BIN file which is discovered in getMassstoragePath before the other disk (where FIRMWARE.BIN and EEPROM.BIN are)

projectkk2glider, 2015-01-14:
Exactly! Can we limit `getMassstoragePath()` to search only in root of mounted drives?

bsongis, 2015-01-14:
I think yes. Careful to regressions on this ;)

projectkk2glider, 2015-01-14:
getMassstoragePath() issue moved to #2004.

projectkk2glider, 2015-01-14:
The `FindMassstoragePath()` is already limited to root of SD card, issue #2004 closed. 

Perhaps he has FIRMWARE.BIN in root directory of some other mounted device (might as well be Taranis SD card). But how can we distinguish which device is the right one in such case?

jtaylor2, 2015-01-14:
|projectkk2glider:
|Why do you have such a weird path to Taranis firmware file /mnt/taranis/eeprom/FIRMWARE.BIN? |Can you show the output of mount when radio is connected in Bootlader mode.

That's just where I chose to mount it.  I don't use automount and mount manually.  I have a script to mount it.  Output from mount (I deleted all the output except the disks)

ThinkCentre:~/frsky/opentx # mount
/dev/sda7 on / type ext3 (rw,relatime,data=ordered)
/dev/sda5 on /boot type ext2 (rw,relatime)
/dev/sdb1 on /mnt/taranis/sdcard type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/sdc on /mnt/taranis/eeprom type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)

|bsongis:
|I try to guess ;) He has an eeprom directory on his SD (labeled taranis) and in this folder there is a |FIRMWARE.BIN file which is discovered in getMassstoragePath before the other disk (where |FIRMWARE.BIN and EEPROM.BIN are)

There is no FIRMWARE.BIN or eeprom directory on my sdcard.  There are several FIRMWARE.BIN files on my hard drive and the system has many eeprom directories, but none of the eeprom directories cantain a FIRMWARE.BIN except the one on the mounted Taranis.  See find outputs below:

ThinkCentre:/ # find / -name FIRMWARE.BIN
/mnt/taranis/eeprom/FIRMWARE.BIN
/root/frsky/companion/2.0.2/FIRMWARE.BIN
/root/frsky/companion/work/FIRMWARE.BIN

ThinkCentre:/ # find / -name eeprom
/lib/modules/3.11.10-25-default/kernel/drivers/misc/eeprom
/lib/modules/3.16.7-7-desktop/kernel/drivers/misc/eeprom
/lib/modules/3.11.10-21-default/kernel/drivers/misc/eeprom
/lib/modules/3.11.10-21-desktop/kernel/drivers/misc/eeprom
/lib/modules/3.11.10-25-desktop/kernel/drivers/misc/eeprom
/lib/modules/3.16.7-7-default/kernel/drivers/misc/eeprom
/usr/src/linux-3.16.7-7/Documentation/misc-devices/eeprom
/usr/src/linux-3.16.7-7/drivers/misc/eeprom
/usr/src/linux-3.11.10-21/Documentation/misc-devices/eeprom
/usr/src/linux-3.11.10-21/drivers/misc/eeprom
/usr/src/linux-3.11.10-25/Documentation/misc-devices/eeprom
/usr/src/linux-3.11.10-25/drivers/misc/eeprom
/usr/src/linux-3.11.10-25-obj/i386/desktop/include/config/eeprom
/usr/src/linux-3.11.10-25-obj/i386/default/include/config/eeprom
/usr/src/linux-3.16.7-7-obj/i386/desktop/include/config/eeprom
/usr/src/linux-3.16.7-7-obj/i386/default/include/config/eeprom
/usr/src/linux-3.11.10-21-obj/i386/desktop/include/config/eeprom
/usr/src/linux-3.11.10-21-obj/i386/default/include/config/eeprom
/mnt/taranis/eeprom
/root/frsky/taranis/eeprom

projectkk2glider, 2015-01-14:
Ok, so it is not a wrong dir problem. Are you cleanly unmounting Taranis, before you remove the USB cable?

jtaylor2, 2015-01-14:
Yes, I always umount.  Keep in mind I have flashed every version of openTX 2.x this way using self compiled Companions and they all worked until 2.0.14.

Thanks for looking into it, but don't waste your time on it, I have other ways to flash.  If somebody else had the problem it would be different.  I think I'll compile next and see if it hangs too.

projectkk2glider, 2015-01-14:
We need to resolve your problem, because it might affect other users in Linux platforms. Looks like your are quite handy in Linux and I want to use this opportunity to debug this issue. I will send you new instructions in next couple of minutes for you to try.

projectkk2glider, 2015-01-14:
Here is what I want you to do: I want to see the FAT table or the radio before and after you write the firmware. To do this follow this instructions:

* start Taranis into Bootloader and connect it to computer
* dump first 8 sectors of the device which contains FIRMWARE.BIN like this (replace /dev/sdc by actual device name):
```
# dd if=/dev/sdc count=8 > before.fat
```

* Then do the firmware upgrade in Companion (you can burn the same firmware as it is currently on the radio if you want).

* then repeat the above command, but this time in another file:
```
# dd if=/dev/sdc count=8 > after.fat
```
And then upload those files (before.fat and after.fat )somewhere and post links here. Thanks.

jtaylor2, 2015-01-14:
Compiled next and it fails the same way (hang and can't power off).

Requested files:

before
http://s000.tinyupload.com/index.php?file_id=98175475222447695708

after
http://s000.tinyupload.com/index.php?file_id=92314357194872579299

davxlw, 2015-01-14:
Just to add my experience, self built Companion TAG 2.0.15 on Mint, and no problem flashing the TX (Taranis B) either ON or OFF...

jtaylor2, 2015-01-14:
Just as a sanity check and to make sure nothing had changed in my environment I just checked out and compiled  2.0.12 and used Companion 2.0.12 it to flash 2.0.15 in mass storage mode and it flashed with no problem.  Something about my system doesn't like whatever changed in 2.0.14.

projectkk2glider, 2015-01-15:
I've analyzed your FAT files and everything appears OK. So far we eliminated:
* wrong file found by `findMassstoragePath()`
* FAT cluster rearrangement
* improper un-mounting of radio

I am currently out of ideas what could be wrong here. But here are some ideas to explore:
* add a couple of flash read/write sector counters in Bootloader and display them on LCD, also add some kind of status display
* change copy method of Companion (add delays between sectors, use direct IO if possible, etc..)

@bsongis This appears to be a new type of issue with firmware/eeprom writing related to #1793. I am afraid that the current method with *fake* drive will always have a potential for things to go wrong. I am thinking of how to improve this situation while not changing the procedure from the users point of view.

kilrah, 2015-01-15:
>  I am afraid that the current method with fake drive will always have a potential for things to go wrong.

To be honest I've always been surprised it worked at all. I pointed out all the problems we're having now as potential issues when Mike chose to go the virtual drive path. As usual, the "see, it works!" won, even if it often means it will break later as it's based on unspecified behavior. Sigh.

bsongis, 2015-01-15:
Agree with @kilrah here. I think the current method is not the good choice. Any proposal for a new protocol we could use here? This would be more work, but I would learn something new and that's not a bad thing :)

jtaylor2, 2015-01-15:
> I am currently out of ideas what could be wrong here. But here are some ideas to explore:
>    add a couple of flash read/write sector counters in Bootloader and display them on LCD, also add > [...]

This got me thinking, I don't know the code and it has been 40 years since I did any programming but I thought I would do some experimenting.  Using 2.0.15 it looked like doCopy in avroutputdialog.cpp was doing the writting so I tried slowing it down.  First I tried a loop, which slowed it down, but it still didn't work.  Then I slowed it down a lot using sleep(1).  It took 10 minutes to write the flash, but it worked.  I flashed three times with different code and it worked every time.  I don't know if my loop just wasn't enough delay or if sleep() is doing something else to make it work.

Throwing it out for what its worth.  Here is what I changed:

  if (source.open(QIODevice::ReadOnly)) {
    QFile dest(destFile);
    if (dest.open(QIODevice::ReadWrite)) {
      addText(tr("Writing file: "));
      for (int i=0; i<blocks; i++) {
//      for (int t=0; t<10000; t++) {
//              printf("loop %i /n",t);
//      }
        sleep(1);                                                         //this makes it work, but takes 10 minutes to flash
        int read = source.read(buf, BLKSIZE);
        if (dest.write(buf, read) == read) {
          dest.flush();
          ui->progressBar->setValue(i);
          if ((i%2) != 0)
            addText("#");
                                                  153,17        27%

kilrah, 2015-01-15:
> Any proposal for a new protocol we could use here? This would be more work, but I would learn something new and that's not a bad thing :)

https://github.com/opentx/opentx/issues/2008#issuecomment-70058231

Kennyptho, 2015-04-21:
DFU UTIL EXE. NOT FOUND ... WHAT DO I DO? HELP!!!

projectkk2glider, 2015-04-21:
@Kennyptho please use forums for such questions. This site is for actual bugs/enhancements to opentx.

tomkfly, 2015-04-21:
Go to http://open-txu.org. If you have any questions, use the Forums and supply full information of equipment being used, firmware and companion version.

closed 2015-04-21 by bsongis

gokl, 2015-05-07:
Hi,

please reopen this issue. I'm also having the exact same issues @jtaylor2 is experiencing.

One needs to check "Check Firmware compatibility" for the flash to fail silently.

To me it looks like the problem is in avrOutputDialog::doCopy() / CopyProcess::onTimer(). 
Specifically its the different QIODevice::OpenMode's (QIODevice::ReadOnly for checking firmware compatibility vs. QIODevice::ReadWrite for flashing the update for the FIRMWARE.BIN file).

When changing the QIODevice::ReadWrite for writing the firmware to QIODevice::WriteOnly its working.
Also when changing QIODevice::ReadOnly to QIODevice::ReadWrite for reading the FIRMWARE.BIN for the compatibility check the later firmware write with QIODevice::ReadWrite is working fine !!

As i'm not into QT i don't quite understand the relation here.
FIMRWARE.BIN is being opened for reading only to check the compatibility and right after that the file is being closed.
Later, when writing the update, the same file is being opened in read write mode and now that's a problem. Opening here in write only mode works fine.

Could someone explain this to me? And maybe fix this strange behavior by using QIODevice::ReadWrite for both file access or only QIODevice::ReadOnly and QIODevice::WriteOnly (as the file is only being written to).

Thanks

kilrah, 2015-05-08:
What are your platform details? Can't see a problem here on Windows.

Kennyptho, 2015-05-08:
I am running windows & home. AMD Phenom(tm)9950 Quad Core Processor 2.60GHz
installed memory (RAM) 8.00 GB, system type: 64-bit OS
My Taranis is a X9D. 017523B01, In the empty Modgual compartment the tag says: 001 AU MWC and the date says: 02/06/2014.

gokl, 2015-05-08:
I'm on openSUSE Tumbleweed x86_64, Kernel 4.0.1-1-desktop. The partition is mounted like this

/dev/sdc on /run/media/user/TARANIS type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=100,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)

Kennyptho, 2015-05-08:
lol thats all french to me sorry

kilrah, 2015-05-08:
> I am running windows 7 home [...]

The question was for @gokl.
Your issue has absolutely nothing to do with this, and is just you needing to configure dfu-util path as you were already told in another unrelated issue's comments.

jtaylor2, 2015-05-08:
@gokl I tried changing  (dest.open(QIODevice::ReadWrite)) to WriteOnly in doCopy (is that where you meant) and it did NOT fix my problem.  I compiled 05/08 master Companion with the change (see diff below) on 32 bit OpenSUSE 13.2.  Write firmware (downloaded 2.0.17) in massstore mode completed with no errors, unmounted Taranis and when I hit exit it hung with backlight on, and wouldn't power off just like all the other failures I've been getting on OpenSuse since 2.0.14.  Pulled the battery to power off and flashed the same firmware with DFU to recover. 

```
ThinkCentre:~/frsky/opentx/git/opentx/companion/src # git diff origin/master
diff --git a/companion/src/avroutputdialog.cpp b/companion/src/avroutputdialog.cpp
index 8fe35b8..fd3862e 100644
--- a/companion/src/avroutputdialog.cpp
+++ b/companion/src/avroutputdialog.cpp
@@ -138,7 +138,7 @@ void avrOutputDialog::doCopy()
 
   if (source.open(QIODevice::ReadOnly)) {
     QFile dest(destFile);
-    if (dest.open(QIODevice::ReadWrite)) {
+    if (dest.open(QIODevice::WriteOnly)) {
       addText(tr("Writing file: "));
       for (int i=0; i<blocks; i++) {
         int read = source.read(buf, BLKSIZE);
```

gokl, 2015-05-08:
@jtaylor2  yes that worked for me. Did you also try to give both file accesses readwrite permissions? That did work for me, too. As i'm on holiday for a week now i wont be able to do more testing..

jtaylor2, 2015-05-08:
@gokl  I will be leaving tomorrow for 10 days with only a Windows 8.1 tablet and no time so won't look at it any more until I get back either.

PR #1998 (2015-01-13): Fixes #1997: firmware upload failed when "Check Firmware Compatibility" ...
...was selected

Closes #1997 

I've tested this on Linux with *fake* Taranis - just an USB key with FIRMWARE.BIN file. But it works - the contents of FIRMWARE.BIN get changed to desired state.

We need somebody to test this on Windows and with real radio.

The diff is not very nice, because I also changed indentation.
commit 0271823d (2015-01-13): Merge pull request #1998 from opentx/projectkk2glider/issue_1997

Fixes #1997: firmware upload failed when "Check Firmware Compatibility" ...
commit d8273954 (2015-01-13): Fixes #1997: firmware upload failed when "Check Firmware Compatibility" was selected
commit fcbe4572 (2015-01-14): Fixes #1997: firmware upload failed when "Check Firmware Compatibility" was selected (ported from master)