opentx/opentx #2001: Modifying SPORT parameters using the Taranis

fraternl, 2015-01-14:
Mike Bland made it possible to use the Taranis to upgrade the firmware of the SPORT devices. Frsky has Windows software to change parameters like sample rate (the one I'm interested in).
I have already ordered the USB-stick so I can use the Windows software for this, but it would be nice if we in the future could do it with the Taranis.

Mike was interested in doing this, but he needs to find time.

kilrah, 2015-01-15:
Duplicate of #1599

fraternl, 2015-01-16:
Hi Kilrah...
This is not a duplicate of 1599
It's updating the parameters of the SPORT-devices.
Not updating the firmware.
Mike wasn't even aware until a few days ago.
He got in touch with Frsky that gave him the info to do this.... one day.

kilrah, 2016-06-14:
#3461

griebd, 2016-06-14:
nice feature in my opinion, why close it? hope it was a miss click while commenting... reopening it! If I'm wrong, please explain why it's not going to happen...
p.s.: IF and WHEN I have the time I will look at implementing it, if not already implemented...

griebd, 2016-06-14:
sorry, my mistake! now I see you moved it to #3461...

closed 2016-06-14 by griebd

bsongis, 2016-06-15:
It's already done