opentx/opentx #2006: A 2nd "Instant trim" that resets the trims and moves them to "subtrim"

fraternl, 2015-01-14:
The instant trim as it is implemented now is still useful.
I would however prefer to trim my model using the trim buttons and when I'm satisfied I want to reset all the trims of that flight mode.
The status of the sticks should be discarded.

This could be done during flight using a special function.

I now use the "servos page" for this.

davxlw, 2015-01-15:
If you scroll down the list in the SERVOS page, you'll see the function Trims => Subtrims which does all servos at once (long press [ENT]):
[image]

kilrah, 2015-01-15:
Duplicate of #1952

closed 2015-01-15 by kilrah

fraternl, 2015-01-15:
@davxlw I didn't know about that option and will use that instead of each servo. I assume I need to manually set the trims to 0 afterwards?

But it wasn't what I ment, really
I want that to be accessible using a SF together with the reset to 0.
It could then happen in flight.
Apparently someone else thought of that one already...

davxlw, 2015-01-15:
@fraternl,
Don't hesitate to try the options on the simulator :wink: 
And no, you don't need to set trims to zero, Trims => Subtrims takes care of that.