opentx/opentx #2008: Possibility to use Taranis serial port to send/receive menu items to PC/Linux/Ubuntu

geekness, 2015-01-15:
I'd like to know if it's possible to use the serial port on the Taranis to view a real time menu on an Ubuntu system?
I understand that I would need an interface on the Ubuntu side, which is easy enough.

I have seen that the serial port outputs telemetry, which is a start.

geekness, 2015-01-15:
Note: I don't necessarily need to use the serial port, any suggestions from firmware developers will be investigated.

kilrah, 2015-01-15:
See the "Hardware" menu, Serial Port item, everything that's implemented is there. There's no such function at this point. 

We do wish to one day have a way to do "live" edits of radio settings through a data link between companion and the radio though. That way companion or an instance of the simulator could edit settings that are in RAM on the radio in real time instead of through a complete EEPROM transfer and model reload, and by extension it could also generate the display itself. 

Kind of very low priority at the moment though. If you want to help, you're welcome. 

What I had in mind was to develop a protocol between the 2 so that:
- The radio sends its physical inputs (switches, pots, sticks, trainer etc) to the PC, and the PC sends its own inputs (joystick, GUI "sticks, switches and pots" etc) to the radio on a regular basis.
- The settings (basically the model data and general settings structs) are synced anytime they are changed, and on a regular basis as safety.

That could run over USB HID, serial or both.

pindeslandes, 2015-01-15:
What about just plugging directly a keyboard to the Taranis to make it easier to enter numbers and characters? I have no idea whether it would be simpler to implement or not but if yes, it might be useful (and usable at the field too).

kilrah, 2015-01-15:
That's not what he's after, it would be a potentially even larger amount of work, and the hardware doesn't even support it without dirty hacks.

geekness, 2015-01-15:
Thanks for the replies.
What kilrah is suggesting sounds like what I'd like to implement.
Im not a programmer, im more of a hacker. I'd like to help out any way I can, so let me know what I can do and I'll try my best. Just give me direction and im generally good after that - Google is my friend.

SamuelBrucksch, 2015-01-25:
I would suggest to add something, that you can use LUA to send custom data over the serial port. I want to create a telemetry screen that already calculates some stuff from the telemetry values and put it out on the serial port in my own format. This would be pretty cool.

projectkk2glider, 2015-02-21:
If we choose to implement this via serial interface, the we can look at the http://stm32f4-discovery.com/2014/08/library-24-virtual-com-port-vcp-stm32f4xx/ (Virtual COM Port (VCP) for STM32F4)

bsongis, 2015-02-21:
Thanks for this link this is really interesting!

projectkk2glider, 2015-02-21:
Look also at other libs for STM32F4

geekness, 2017-01-13:
I know it's an old thread, but any idea if this feature has progressed at all?
Same as this thread, so they could probably be merged?
https://github.com/opentx/opentx/issues/2604
This one has a milestone of 2.1.X, anybody reckon it's possible to make that milestone?

kilrah, 2017-01-13:
A CLI has been implemented on the serial port, but the only commands available there at this point are for debug purposes.

closed 2019-11-15 by stale[bot]