opentx/opentx #2009: Timer Voice Miniute Call not working on  V2.0.15

tomkfly, 2015-01-15:
Checked my TX after seeing this reported on RC Groups and the mins are not being called out.

Tom

kilrah, 2015-01-15:
Works fine here. More details?

tomkfly, 2015-01-15:
I have my timer 1 set to  THt, 20 mins , persistant  flight, Minute call  checked and Countdown Voice,
With V2.0.13 it used to call out the mins 19,18, 17 --- down to 1 and at then count seconds down.
Since changing to V2.0.15  I am no longer getting these call outs.
I have checked the sounds are still in the SD Card and will call out if I set up a SF to playe Value off a switch.

projectkk2glider, 2015-01-15:
Upload the eepe with the model that does this somewhere and post link here.

kilrah, 2015-01-15:
OK, confirmed on a new model with same settings.

projectkk2glider, 2015-01-15:
Maybe fixes in #1927 caused this one?

projectkk2glider, 2015-01-15:
Yep, my commit in #1927 caused this. Damn, you fix one thing and brake another :). Fix coming...

commit 8c73df5e (2015-01-28): Fixes #2009: Timer Voice minute call not working for persistent timers
Timers moved to timer.cpp
Added timer overflow protection
Added timer gtests
THt mode now triggers on THR>10%
commit 9e501031 (2015-01-28): Merge pull request #2021 from opentx/projectkk2glider/issue_2009_timers_refactoring

Fixes #2009: Timer Voice minute call not working for persistent timers
PR #2048 (2015-01-29): Fixes #2009: Timer Voice minute call not working for persistent timers
Timers moved to timer.cpp
Added timer overflow protection
Added timer gtests
THt mode now triggers on THR>10%

Closes #2009 

Same as #2021, but for `next` branch.
commit afc41f08 (2015-01-29): Fixes #2009: Timer Voice minute call not working for persistent timers
Timers moved to timer.cpp
Added timer overflow protection
Added timer gtests
THt mode now triggers on THR>10%
closed 2015-01-29 by bsongis