opentx/opentx #2010: SB middle position not work

laflaf3d, 2015-01-15:
Hi,
This morning I tried to use the middle position of SB switch in simulation. It's doesn't work.
In the same hand : The NOT SB in HIGH position is set only when SB is in LOW position.
[image]

tomkfly, 2015-01-15:
You should have !SB^ and SB^. As your switches

projectkk2glider, 2015-01-15:
And don't use the same switch for **mixer source** and **mixer switch** unless your really know what you are doing. You might want to use **MAX** for mixer source in your case and perhaps adjust the weight of some mixes to -100%.

Closing as this is user error, not opentx issue.

closed 2015-01-15 by projectkk2glider

laflaf3d, 2015-01-15:
Shame on me !Your great, thanks a lot for your marvellous work !
Stéphane. 

   

 Closed #2010.—