opentx/opentx #2015: Bad antenna after flight or telemetry reset

janekx, 2015-01-16:
Did found bad bug in reset of the Flight and Telemetry. 
Taranis Plus B01 
When reset flight or telemetry, the SWR value goes from 0 to 255 and back to 0, in this time OpenTX alert bad antenna.
This should not happend.

projectkk2glider, 2015-01-16:
Confirmed from the sources:
* when telemetry is reset all values go to zero (including )
* until XJT_VERSION_ID is recieved the `IS_VALID_XJT_VERSION()` is false
* if SWR packet comes and `IS_VALID_XJT_VERSION()` is still false, the SWR gets set to 0xFF.

bsongis, 2015-01-28:
Confirmed. I can't remind why we set SWR to 0xFF when IS_VALID_XJT_VERSION() is false. Right now I would set it to 0x00

bsongis, 2015-01-28:
I will mark this one as fixed once the whole SWR masking feature is ported to 'next'

commit 013de888 (2015-01-28): Fixes #2015
commit 15378b8f (2015-01-28): Fixes #2015
closed 2015-01-28 by bsongis