opentx/opentx #2016: Switch position warning on startup

ululi1970, 2015-01-16:
After upgrading to 2.0.15 from 2.0.13, I have an issue with the warning of POT positions. Before the upgrade, the warning windows appeared if a POT was not on the last recorded position, with an arrow to indicate in which direction to rotate the POT to go back to the last recorded position. Once the POT was 
set right, the warning disappeared. Now, it does not seem to work. I can rotate the POT back and fort and the only thing I get is a beep when it goes past center.

projectkk2glider, 2015-01-17:
I can not reproduce this,  tried every possible case, it always works for me. @ululi1970 any chance you screwed your sticks/pots calibration data?

davxlw, 2015-01-17:
No problem here either...

@ululi1970,
You could post an eepe.

Taranis B
OpenTX 2.0.15

ululi1970, 2015-01-17:
MY mistake. It works as intended. Sorry for increasing your work load. BTW, the fix for blank screen on cold starts works without a hitch...

closed 2015-01-17 by bsongis