opentx/opentx #2017: Copy Min, Max and PPM center of a channel to all the other ones

pindeslandes, 2015-01-17:
**Typical situation:**
You want to replace your radio by a Taranis but you want to have your models settings unchanged. The problem is that the coding of the servo positions varies from a brand to another:
OpenTx = 1500 µs +/- 512 µs
Futaba = 1520 µs +/- 420 µs
JR = 1500 µs +/- 400 µs
Etc.
So you have to modify the center and throw of every channel in the SERVOS page of all your models. But it would be more convenient if you could set those default parameters for all the channels in a row.

**Solution:**
In this pop up menu you get when long pressing ENT in the SERVOS page…  
[image]
…add a **‘Align Min, Max and PPM centers’** line to copy those current channel’s settings to all the others.

ckleanth, 2015-01-17:
this has nothing to do with radios but it is true that every servo manufacturer uses different servo centres.
one can use more than one servo manufacturer in a model.

kilrah, 2015-01-17:
I don't see the point. Those values need to be set to match the menchanical setup anyway, so no 2 servos will typically have the same values.

pindeslandes, 2015-01-18:
OK, I wasn’t clear enough. I take an example:
- You already have a model that you use to control with a Futaba transmitter.
- Whatever the servo’s centers you have in your model (you generaly even don't know if it is 1500 µs or 1600 µs or anything else), you have done your settings so that the neutral positions of the control surfaces match the center position of the transmitter’s sticks, of course.
- That means that the control surfaces are at neutral when their servos receive a 1520 µs order.
- The same for the throws: 420 µs correspond to the 100% throws you want on your model.

Now if you replace the Futaba by a Taranis, the servos will receive a 1500 µs order when sticks are at neutral, so that they will offset by 20 µs (and a 512/420 ratio will be applied to the throws too).
That’s why you have to apply the same correction to all the channels to recover the initial settings of your model.

kilrah, 2015-01-18:
And my point is that in your model setup on your Futaba each servo will likely already have different and non-zero values for subtrim/end points. So you'll have to redo the settings from scratch to get your setup right anyway.

projectkk2glider, 2015-01-18:
I agree with @kilrah you have to do the whole servo limits and sub-trim every time you change transmitters and there is no easy way out. I vote for closing this issue.

pindeslandes, 2015-01-19:
Well, depending on the user, it would do between 0% and 100% of the job!
In my case, almost 100% of the job because I generally try to avoid using subtrims (because most of the time, the manual doesn’t tell you exactly what does a subtrim, which makes me suspicious, and I find it better to focus on the linkages (but I love the fact that I can choose how a subtrim behave on OpenTx)) and I rarely need to move the end points from 100% (most often just for throttle).
I thought I wasn’t the only one in that case and that this enhancement might be helpful. But if I am an exception, well, too bad for me!

ckleanth, 2015-01-19:
Phil,  have a read
http://www.rcgroups.com/forums/showpost.php?p=27079455&postcount=44294
And you will understand why why you should start using subtrims.

kilrah, 2015-01-20:
Or, in the case of opentx, PPM center.

ckleanth, 2015-01-20:
yes but there is a settle difference between using subtrim and modyfying PPM centre
this is what we have in the manual (and I'm probably quoting stuff you wrote)
Subtrim behavior: When set to default, adjusting subtrim will only shift the center of the servo throw. Given a -100% to +100% order from the mixer, the servo will still move exactly between the lower and upper limits, without clipping or dead band. This introduces a different stick to servo movement relation for both sides of the stick. Depending on the situation it can be either convenient or problematic, so the = setting has been added to change subtrim effect to rather shift the servo throw "symmetrically". A full throw order from the mixer can now be clipped by the limit that is on the same side as the subtrim, while on the other side the servo will not reach the limit anymore. That way on both sides of the stick a given stick movement always results in the same servo movement. Typically using the default mode allows for faster setup of servos that are driven by a single control input, while = is required to keep correct response of control surfaces using differential and/or mixing several inputs together. The = mode typically requires reducing D/R so that a margin is left between full "control" throw and the defined limits.

The following diagram illustrates the respective behavior of both subtrim modes and how Center adjustment compares to them:
[image]

kilrah, 2015-01-21:
Yup, that's precisely why you want to use PPM center to center your horns as I said above. 
If you used subtrim, you'd already be introducing dissymmetry (sp?) before you even start doing anything.

ckleanth, 2015-01-21:
Andre, just a question for something I should had noticed long before. I thought once you set your servo end limits, these are hard limts and can never be overidden event he overide chX function. From the plot it appears to suggest that if you use PPM center to center the horns this means that the hard limts are overidden. I thought the effect using PPM whould be similar to subtrim with sym option on.??

kilrah, 2015-01-21:
PPM center shifts EVERYTHING including limits. It's the first thing you should set to get your horn where you want it before you do anything else, and you don't touch it afterwards.

ckleanth, 2015-01-21:
learned something new today I guess I have a few planes to sort out.
thanks Andre.

pindeslandes, 2015-01-21:
ckleanth, thank you for the link but things are actually a little more complicated because pushrod at 90° to the servo arm gives you symmetrical throws only if the length of the servo arm equals the length of the control surface horn.
Let’s take a rigid pushrod, not a sleeved one (it is easier to figure out how it works), with the lever of the control horn longer than the servo’s one (as it is most of the time), let’s say twice as long.
If the servo arm is shorter than the control horn, the distance between the two articulations at both end of the pushrod shorten when the servo arm moves from the middle position, giving an amplifying effect when the servo arms moves away from the control horn and a reducing effect in the other way. In that case (1/2 ratio between the two levers), the best angle is more like 83°.
And, by the way, you also have a positive exponential effect (about 9% more sensitive around the center, here, if the servo moves +/- 45°)!
And yes, I do use ppm and subtrim when mechanics is not enough, as long as I understand what I am doing (which fortunately is the case with the Taranis but not on every other radio).

mizuta1951, 2015-07-09:
Hello pindeslandes,

I had the same kind of issues.
Maybe most of the people does not consider about the PPM center and end points.
In my case I had more than 10 planes stored in the hangar located at the RC field.
When I switched from Futaba to Taranis I had no plane at home, I only had two transmitter and some receivers.
Trims and end points were all adjusted and I did not want to do any other adjustments and an additional maiden flight for them.

The solution that I found was,
1.	Use Hitec HFP-25 servo programmer (this can measure the output pulse of the Rx) and record all neutral value and end points.(if you use D/R then also its value)
2.	Program Taranis mixes and others (this is the thing that you cannot bypass).
3.	At Taranis “Servo” menu, input PPM center to the value according to “1.”.
4.	Also adjust the limit to match the value “1.” (some case you need to change the servo direction).
5.	You can monitor the pulse value on the Taranis Servo menu and confirm it is correct.
6.	Finally don’t forget to check all the control surfaces are moving correctly before your first flight with Taranis.

Additional information,
Futaba’s end point which normally said 100% = +/- 420us is NOT the hard limit.
With mix like delta-mix,V-tail,etc, the servo will stroke up to 135%(typical on 14MZ).
So be careful when you convert them.

I understand when converting the transmitter, the difference of PPM center and limits are really a big issue.
But in my experience, once you finish all the conversion on you existing planes, it is over, the issue is no more an issue..
And as I did, it is possible to do the conversion without any mechanical adjustments.

One thing I wish to inform, Futaba’s 1520us neutral is not a “lonely wolf”.
Most of the gyros default neutral is 1520us.
And many potential OpenTX users might be using Futaba.

closed 2019-11-15 by stale[bot]