opentx/opentx #2020: [2.0.15] Audio stuttering

LapinFou, 2015-01-17:
Sometime, when I start by Taranis Plus (A01 batch), the announce audio message stutters. This also happen on the field when timer/altitude are played.
Sorry I cannot help you much to reproduce the issue. However I opened a ticket to see if other people also got the same problem.
I also tried on the Taranis Plus (A01 batch) of my soon and got the same issue. This is the worst bug, since it appends only sometime for no apparent reasons.

bsongis, 2015-01-17:
Which release?

bsongis, 2015-01-17:
Did you try an erase of your SD?

LapinFou, 2015-01-17:
No. I will try, however when I received my Taranis+ last year, it was one of them with bad SD card. So I bought a Samsung EVO card (special extreme conditions) and never has any more lost of Audio (SD card not recognized problem). I'm always ejecting it before unplugging it from my PC.
The problem started just after I updates from 2.0.13 to 2.0.15. So I definitely thinking more for a software issue than a SD card.
Anyway, I will post a new message as soon as I will have erase my SD card.

projectkk2glider, 2015-01-17:
My radio is quite well behaved in this regard recently with newer OpenTX versions, but had this issue before. Related issues are #1779, #894 and #894.

LapinFou, 2015-01-17:
Here is a way to reproduce the problem. NB: As explained below, it does not happen systematically. 
Browse the SD card select and play some long audio message. You must play many times the audio file before be able to reproduce it.
I successfully reproduce the stuttering issue on my Taranis+ and also and my son's one.

janekx, 2015-01-18:
Did try to reproduce on T+ A01 without success, The message I try to play takes about 4 sec and tryed it more then 15times. 
Using new Kingston 4GB SD card class4 newly slowformated, try to slowformat SD and copye files back.

projectkk2glider, 2015-01-18:
This is one of the issues that is really hard to reliably reproduce, hence very difficult to debug.

LapinFou, 2015-01-18:
Thanks Jan for trying to reproduce. Indeed it is an issue very hard to catch. I did format the SD card with SD Formatter 4.0. But I still got the problem. Sometimes it is working well for 30min, sometime it happens in 2 min. I'm trying to determine a testcase to reproduce systematically the problem with no luck yet... :(

projectkk2glider, 2015-01-25:
@LapinFou when audio stutters:
* do you have SD telemetry log enabled (running)?
* any Lua scripts active?

LapinFou, 2015-01-25:
No LUA script, no logging. However I assigned the volume to S1.
I'm still looking for a scenario in order to reproduce the problem.

LapinFou, 2015-02-16:
I spend sometimes on my 2 Taranis+ yesterday (speaker upgrade), when playing some WAVs, I got again the audio stuttering issue. But unfortunately, I did not find any scenario to reproduce the problem :(
In order to add more information, I'm using the firmware with those settings:
[url=http://www.servimg.com/view/15931330/196][img]http://i38.servimg.com/u/f38/15/93/13/30/tarani11.jpg[/img][/url]

projectkk2glider, 2015-02-16:
Thanks for th eupdate. This issue is really hard to debug, I hope you come up with a reliable way to reproduce this problem. Then we'll have a chance to fix it.

LapinFou, 2015-02-16:
For sure, this kind of issue are the worst ones...
I will update as soon as I found a scenario, but I'm not very confidant. :(

LapinFou, 2015-04-03:
I was using my taranis last night with Phoenix RX (a good RC simulator). When I trimmed the elevator, the voice announcement 'trim centered' stuttered once. :(
The model memory for the simulator is very very basic. No L, no SF, no complex mix, not sound expect the default one.
Never happened again in the following 1/2 hour. Grrrr still no scenario to reproduce the problem.
The only common point between all models memories is that I'm using a permanent timer.

projectkk2glider, 2015-04-03:
My observations regarding this:
* audio stutters more if SD logging is enabled.
* stutter is usually right after the beginning of the audio file.

Therefore I predict that this issue can be improved filling (for each new file to be played) more than one audio buffer (let say at least three buffers) before starting the audio playback. @bsongis @kilrah what do you think? Something similar was implemented for simu audio otherwise the playback would also stutter on PC (https://github.com/opentx/opentx/blob/next/radio/src/targets/simu/simpgmspace.cpp#L464).

LapinFou, 2015-04-07:
"stutter is usually right after the beginning of the audio file."
That is absolutely correct in my case.

LapinFou, 2015-04-30:
One more information. I'm using 8Go SD card. Maybe, this is an important point.

LapinFou, 2015-05-15:
Hi folks,

I saw that the new Bootloader 2.0.17 was fixing some issue with SD card larger than 4Go #2154.
So, I decided to format my 8Go SD card after the firmware & bootloader upgrade. Since then, I never had anymore the audio stuttering problem.
I don't know if there is a correlation, but for now, the problem is gone.
I will post a comment if the problem comes back.

bsongis, 2015-05-15:
I mark it as 'fixed' in OpenTX 2.0.17 then, feel free to reopen if it comes again

projectkk2glider, 2015-05-16:
@LapinFou I am afraid that related fix has nothing to do with this issue. Your sudden improvement is more of an coincidence or the fact that different SD card is faster.

LapinFou, 2015-05-16:
@projectkk2glider I'm not surprised of your comment. I was suspecting there was no correlation between this bug fix and my problem. Unfortunately I've not been able to reproduce a testcase to highlight this issue. Anyway, if the problem comes back, I will post a new message.
Thanks to all of you for you time and for taking seriously in account the problem.

erieqc, 2015-06-01:
I just updated to 2.0.17 and mine is stuttering now it never did it before till now.

floaaaaargh, 2015-06-21:
Same sound-issues here. 
I encountered them since 2.0.15 and they are not-reproduceable for me. 
When upgrading to 2.0.15 I also installed a bigger SD-Card (8GB) which I now changed back to the old one for troubleshooting. 

When copying files to the SD-Card on my Mac, the computer creates lots of ghost-files beginning with ._ Maybe the Taranis doensn't like them?!?

LapinFou, 2015-06-22:
I highly recommend you to backup, then format you SD card using this tool:
https://www.sdcard.org/downloads/formatter_4/
Give a try and post your result.

floaaaaargh, 2015-06-22:
Changed my SD-Card back to the smalltalk one, which worked trouble-free with 2.0.13. Removed all stupid ._ files on the Card. 
Still stutters frequently. While logging, stuttering ist much worse. 
Next try is formatting SD-Card as suggested in the post above.

floaaaaargh, 2015-06-22:
Forgot to mention, that I am using 2.0.17 since its release.

floaaaaargh, 2015-06-24:
Short update: I have formatted the SD-Card with the tool listed above. With no postitive effect on the sound-stuttering. 
While the stuttering is mostly encountered with logging active, I reduced the logging interval from 0,5 sec to 2.0 seconds. Did some testing and dindn't encounter stuttering so far.

floaaaaargh, 2015-07-11:
Another Update: Additional to the changes above I deactivated the LUA-Telemetrie Script. The stuttering seems to be reduced, but it still happens frequently.

LapinFou, 2015-08-12:
It happens again last Saturday once.
So formatting the SD card did not fix the problem. Must be something else.
Could it be a scheduling priority problem. Meaning, that an IRQ (or something else) with an higher priority happen during the audio playback? It is just a guts feeling.

LapinFou, 2015-11-09:
I will not update anymore this issue, since I switched both my T+ to 2.1.15
So far, no more audio stuttering. Look likes it is somehow linked to the 2.0 branch.

closed 2015-11-11 by bsongis