opentx/opentx #2022: Logging of Logical Switches

fraternl, 2015-01-17:
I have created many logical switches and it's possible to debug this using the simulator. But this isn't always possible as some realworld 

For instance; I have created a logical switch with a duration of 0.2s to reset my telemetry when I enter launch mode, the absolute height is more than 2m and the vertical speed is less than 0.1 m/s.

I would suspect it to work properly, but sometimes it resets telemetry when it's already launched. It says so, at least.

After flying I opened my logs expecting to see the values of the logical switches and found out they weren't logged.

It would be nice if this is added. These values are boolean so it would fit in 4 bytes

fraternl, 2015-08-22:
This never received any attention.
I needed it today to check if the logic was working as intended

Spainman, 2015-08-22:
Being a switch junky myself I consider it a great idea! It would be a great tool to debug complex sequences.

fraternl, 2015-09-27:
Sometimes my timer gives me a 0-output after a flight.
It would be great if logging of switches could be made optional so I might find the cause of it.
Or is it already?

projectkk2glider, 2016-05-16:
Two issues with current implementation:
* label `LS` is already used by `Left Slider`, I propose `LogSw`
* I propose to store HEX value in log (faster conversion, fixed width)

closed 2016-06-28 by projectkk2glider

vian31, 2016-07-05:
I was wondering if this function might be possible... Very good to see that has been included in the 2.2 milestone.
Superposing LS status with log charts will be very helpfull.

LevYas, 2021-11-10:
This issue is very searchable, so I want to add some comments on that.
1. Logging is implemented, but there's no UI to display that. The log viewer can only display these HEX values. Probably, you can open the log in excel and convert the column using formulas to individual logical switches.
2. Here's a clarification on how you can interpret these HEX values https://www.rcgroups.com/forums/showpost.php?p=47996299&postcount=6833

commit 2a4bc2cf (2016-01-30): Fixes #2022
commit 52bb3586 (2016-06-28): Fixes #2022: Logical switches logged in HEX format with LSW header label
commit 19024795 (2016-06-29): Fixes #2022: Logical switches logged in HEX format with LSW header label
Fixed C compiler warnings about "std=gnu++11" flag
Fixes #194: added logging of TX battery voltage
[Horus] Model move within a category
[Horus] Unused string removed
[Horus] Move models from one category to another
PPMCA option is now ON by default
[Horus] Navigation line by line in COLOR widget settings
[Horus] Custom switch names badly displayed