opentx/opentx #2023: lost functions and get errors

fanuc, 2015-01-18:
I updated my opentx companion last week and now get this error Warning box with
EEPROM saved with these warnings:
-OpenTX on this board doesn't accept this function
-OpenTX on this board doesn't accept this function
-OpenTX on this board doesn't accept this function
-OpenTX on this board doesn't accept this function

Also I have lost logical switches and special function menu bars in companion .

bsongis, 2015-01-18:
Which board?

fanuc, 2015-01-18:
Sorry, It's OpenTX for frsky Taranis 
Radio Boot loader 2.0.8 
opentx-taranis-2.0.15 
date 2015-01-14 
EEPR 216
Running on Ubuntu 14.04
Companion Version Updated to Version 2.0.99,JAN 15 2015 from old Version 2.0.14 JAN 12 2-15

kilrah, 2015-01-18:
2.0.99 is an alpha version of the upcoming 2.1 branch, is only available for testing / debugging, completely unstable and not compatible with the current release version.

I.e. Unless you're a developer and know what you're doing don't use it.

closed 2015-01-18 by kilrah