opentx/opentx #2024: Display any parameter in telemetry screens

pindeslandes, 2015-01-18:
I wish it were possible to display any parameter on a telemetry screen. Not only telemetry values and timers, as it is today, but also a channel value, an input value, a GV, a LS, a stick position etc.
That would be very convenient, for example when you are setting a complex mixer and need to have a comprehensive and simultaneous view of all the parameters involved.
Logically, a second part of this enhancement would probably consist in separating the ‘Screen #’ items from the TELEMETRY page (which would be no longer relevant) and include them in a new CUSTOM SCREEN page.

kilrah, 2015-01-18:
Already implemented for 2.1.

closed 2015-01-18 by kilrah

pindeslandes, 2015-01-19:
Nice!