opentx/opentx #2025: Numbers and Letters Missing in both OpenTX and on Taranis X9D

Durahl, 2015-01-18:
I recently made the jump from 1.x to 2.0.15 with my Taranis X9D and then noticed this "Problem":
[image]
I recall there normally being Numbers and Letters before the Upgrade but now it only shows Boxes and Dashes as if the Font is missing. In the OpenTX Simulator it's solid Boxes ■ while on the Taranis X9D it's just the outlines □

Did I miss to installing something?

kilrah, 2015-01-18:
No, it's been changed to that. Dash = unused, empty box = configured and false, full box = configured and true.

closed 2015-01-18 by kilrah

Durahl, 2015-01-18:
May I ask for the reason behind this Change?

I'll be honest, I didn't even use that particular feature before but from a readability standpoint I find it hard to differentiate a bunch of identically looking boxes from each other as compared to what we had.

Something wrong with Dash = Unused, Letter/Number black on white = false, Letter/Number white on black = true?
Just wondering :)

kilrah, 2015-01-18:
LS are now named by number, no more letters. 2-digit numbers don't fit in a grid like before anymore. 
The current arrangement is in blocks of 5, 10 on a row etc... shouldn't be too hard to locate.

projectkk2glider, 2015-01-18:
Also the number of LS was increased and there had to be a change to accommodate a new number of switches. Funny this has been done long ago and nobody complained until now.

Durahl, 2015-01-18:
Thanks for clarification.

Like I said I only just made the change from Before 2.0 to After 2.0 so it's rather fresh to me and with it looking like a typical unrecognized Font Bug on a regular PC I was kind of worried.

I was also a little bit concerned about the lack of !SF↑ and !SF↓ whereas SF↑ and SF↓ were still available but luckily I found a thread answering the question before opening another thread :)