opentx/opentx #2027: Deleting file on radio does not actually frees clusters

projectkk2glider, 2015-01-18:
I was testing full SD card solution for issue #2007 and found out this. Check of SD card in comp:
```
$ sudo fsck.vfat -a /dev/sdd1
fsck.fat 3.0.26 (2014-03-07)
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Reclaimed 31555 unused clusters (1033994240 bytes) in 2 chains.
Performing changes.
/dev/sdd1: 524 files, 60021/60021 clusters
```

bsongis, 2015-01-20:
That's really annoying, we even call unmount before stopping the radio!

projectkk2glider, 2015-01-22:
The bug was fixed upstream, see http://elm-chan.org/fsw/ff/patches.html We just have to apply the patch.

projectkk2glider, 2015-01-22:
Two patches from above link:
```patch
2. November 26, 2014
-------------------------------------------------------------------------------------------
_FS_NORTC feature does not work properly. To fix this problem, apply following patch to
the ff.c.

--- ff_patch1.c	Thu Nov 13 21:16:07 2014
+++ ff_patch2.c Wed Nov 26 19:42:01 2014
@@ -164,6 +164,9 @@

 /* Timestamp feature */
-#if _FS_NORTC
-#define GET_FATTIME()  ((DWORD)_NORTC_YEAR << 25 | (DWORD)_NORTC_MON << 21 | (DWORD)_NORTC_MDAY << 16)
+#if _FS_NORTC == 1
+#if _NORTC_YEAR < 1980 || _NORTC_YEAR > 2107 || _NORTC_MON < 1 || _NORTC_MON > 12 || _NORTC_MDAY < 1 || _NORTC_MDAY > 31
+#error Invalid _FS_NORTC settings
+#endif
+#define GET_FATTIME()  ((DWORD)(_NORTC_YEAR - 1980) << 25 | (DWORD)_NORTC_MON << 21 | (DWORD)_NORTC
_MDAY << 16)
 #else
 #define GET_FATTIME()  get_fattime()

```

```patch
1. November 13, 2014
-------------------------------------------------------------------------------------------
f_unlink function does not remove cluster chain of the file. It reads to a lost cluster of
the FAT volume. This is added at a bugfix of R0.10c.
To fix this problem, apply following patch to the ff.c.

--- ff.c	Sun Nov  9 15:06:49 2014
+++ ff_patch1.c	Thu Nov 13 21:16:07 2014
@@ -3487,20 +3487,22 @@
 					res = FR_DENIED;		/* Cannot remove R/O object */
 			}
-			if (res == FR_OK && (dir[DIR_Attr] & AM_DIR)) {	/* Is it a sub-dir? */
+			if (res == FR_OK) {
 				dclst = ld_clust(dj.fs, dir);
-				if (!dclst) {
-					res = FR_INT_ERR;
-				} else {					/* Make sure the sub-directory is empty */
-					mem_cpy(&sdj, &dj, sizeof (DIR));
-					sdj.sclust = dclst;
-					res = dir_sdi(&sdj, 2);		/* Exclude dot entries */
-					if (res == FR_OK) {
-						res = dir_read(&sdj, 0);	/* Read an item */
-						if (res == FR_OK		/* Not empty directory */
+				if (dir[DIR_Attr] & AM_DIR) {	/* Is it a sub-dir? */
+					if (!dclst) {
+						res = FR_INT_ERR;
+					} else {					/* Make sure the sub-directory is empty */
+						mem_cpy(&sdj, &dj, sizeof (DIR));
+						sdj.sclust = dclst;
+						res = dir_sdi(&sdj, 2);		/* Exclude dot entries */
+						if (res == FR_OK) {
+							res = dir_read(&sdj, 0);	/* Read an item */
+							if (res == FR_OK		/* Not empty directory */
 #if _FS_RPATH
-						|| dclst == dj.fs->cdir	/* or current directory */
+							|| dclst == dj.fs->cdir	/* or current directory */
 #endif
-						) res = FR_DENIED;
-						if (res == FR_NO_FILE) res = FR_OK;	/* It is empty */
+							) res = FR_DENIED;
+							if (res == FR_NO_FILE) res = FR_OK;	/* It is empty */
+						}
 					}
 				}
```

bsongis, 2015-01-22:
Excellent, would you open a pull request for this? Or commit directly, no problem!

projectkk2glider, 2015-01-23:
Ok, later....

projectkk2glider, 2015-01-23:
Tested on my radio, now the file-system is clean after a file is deleted on the radio.

commit 8b455121 (2015-01-23): Fixes #2027: Deleting file on radio does not actually frees clusters (patch from upstream)
commit 1cccbac5 (2015-01-23): Fixes #2027: Deleting file on radio does not actually frees clusters (patch from upstream)
closed 2015-01-23 by projectkk2glider