opentx/opentx #2030: Open TX companion

ginney54, 2015-01-19:
Running win 7 Dell OpenTX unable to check for updates. Will not download firmware. Works ok on my work Dell running Win8. Headbanger.

kilrah, 2015-01-19:
What companion version? 
If the same version works fine on another computer the problem is likely with the computer.

ginney54, 2015-01-19:
Thanks Andre both have the latest version, must be the computer but can’t get it sorted. I will keep looking. I have flashed the Taranis Plus using the file that downloaded from the working PC. Have looked at firewalls and cleared all cache.
Mystery. I will let you know if I get it fixed.
Greg

 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com> 
Version: 2015.0.5645 / Virus Database: 4260/8954 - Release Date: 01/18/15

projectkk2glider, 2015-02-28:
No feedback, closing.

closed 2015-02-28 by projectkk2glider