opentx/opentx #2031: openTx 2.0.15 Taranis Plus internal XJT and D4R-II problem

er9xuser, 2015-01-19:
I have D4R-II receiver flashed with 27ms FW and Taranis Plus. So have binded Rx to Taranis INTERNAL XJT Tx and binds ok. But when you turn off Tx and Rx...and turn them on again...green light on Rx flashes in strange sequence. And when I've tried to fly quadcopter...it is impossible.

Then I have tried using my old external XJT module. Put it into Taranis and binded Rx with external XJT...this combination works fine.

Last thing I did is I've loaded stock Taranis Plus FW 2.0.9 - FrSky version. And binded Rx with INTERNAL XJT again and this combination works just fine.

projectkk2glider, 2015-01-19:
On the forums you write that you also changed internal XJT firmware to the new EU compatible version.  When you changed back to OpenTX 2.0.9, did you also flashed back the old XJT firmware, or do you now have a working combination with:
* OpenTX 2.0.9
* the new EU firmware on XJT

Because nothing related to RF module was changed in OpenTX since 2.0.9.

er9xuser, 2015-01-19:
No I didn't changed nothing on internal XJT - I just wrote that Taranis I have...it's been made this year and firmware on internal XJT by default complies with new EU ETSI ENE 1.8.1 standard.

projectkk2glider, 2015-01-19:
Hmm, just browsed over to FrSky download site http://www.frsky-rc.com/download/view.php?sort=Firmware&down=168&file=Firmware-Taranis%20X9D%20Plus and seen:
```
Compatiblity  :  Taranis X9D Plus
UpdateTime  :  2014-10-10
Version  :  opentx-x9dp-v2.0.9
File Size  :  527KB
MD5  :  323955BE237FEC51057424D634B4AED8
Total downloads  :  4458

notes:

PXX output signal bug fixed
```

@bsongis  Did FrSky made some change to 2.0.9 (like PXX output signal bug fixed) and forgot to tell us? Also build date is a little late for 2.0.9, we released it on Aug 22, 2014

projectkk2glider, 2015-01-19:
I have also found this version:
```
Die aktuelle FW: opentx-x9dp-v2.0.9.2
Date: 2014-10-28
Time 01:06:17
EEPR: 216
```

er9xuser, 2015-01-19:
projectkk2glider...could you please contact me over rcg PM and send me your phone number....so we can talk and I'll explain this problem better. Thanks.

projectkk2glider, 2015-01-19:
Relevant information from @er9xuser:

>1st scenario:
Taranis Plus rev. B02 - made 8.1.2015 for EU (internal XJT complies with new EU ETSI 1.8.1 standard) - with stock openTx 2.0.9 FW. Binded with D4R-II Rx (this Rx is flashed with 27ms FW) works with no problem - green lite on Rx is solid.

>2nd scenario:
Taranis Plus rev. B02 - made 8.1.2015 for EU (internal XJT complies with new EU ETSI 1.8.1 standard) - with openTx 2.0.15 FW. Binded with D4R-II Rx (this Rx is flashed with 27ms FW) not working - green lite on Rx flashes in strange sequence.

>3rd scenario:
Taranis Plus rev. B02 - made 8.1.2015 for EU using (external XJT which has this FW - http://www.frsky-rc.com/download/vie...e=Firmware-XJT - this doesn't complies to new EU ETSI 1.8.1 standard) - with openTx 2.0.15 FW. Binded with D4R-II Rx (this Rx is flashed with 27ms FW) works with no problem - green lite on Rx is solid.

>4th scenario:
Last thing I did...I've downgraded openTx FW from 2.0.15 to stock Taranis Plus openTx FW 2.0.9 and everything works like in 1st scenario. 

Again, I think FrSky changed something in their OpenTX 2.0.9 and is now different from our 2.0.9 version. @er9xuser tell us what is displayed on your radio version (Radio settings, Version page) when you have original FrSky 2.0.9 version flashed (we want version and build date).

projectkk2glider, 2015-01-19:
I've talked with @er9xuser and his problem is:
* OpenTX 2.0.15
* internal XJT with EU firmware (his radio came stock with this firmware)
* using D8 protocol
* D4R-II Rx receiver with  27ms firmware flashed

er9xuser, 2015-01-19:
I'll post what is displayed under Version page tonight.

er9xuser, 2015-01-19:
Info from Version screen:

VERS: opentx-x9dp-v2.0.9
DATE: 2014-08-05
TIME: 11:04:32
EEPR: 216

kilrah, 2015-01-19:
> internal XJT with EU firmware (his radio came stock with this firmware)

Is that really sure? The firmware came out 2 weeks ago, I doubt any have been delivered with it yet unless you really just bought it a couple of days ago and your dealer did the upgrade himself.

> using D8 protocol

From the info we have D8 has not change from one firmware to the other, only D16 has.

er9xuser, 2015-01-19:
Yes. Got Tx last friday 16.1.2015. In the external JR bay, there is a sticker with etsi ene 1.8.1 writteb. And whe I've opened Tx because I like my throtle stick a little bit harder....inside on back pcb where internal xjt module is written EU. So yes, frsky already flashed tx module with eu fw. Something is not ok with 2.0.15 firmware.

projectkk2glider, 2015-01-20:
We have an answer from FrSky:

>There are 3 verion of plus firmwares RELEASED differing in compile date.
1.0805,the first release version,hide the SWR compared to X9D firm
2.1010,to fix the wrong PXX output in version one(caused by newer GCC version).
3.1028,to enable SWR.

All those changes are also in OpenTX so my theory about FrSky's version being different is wrong. 

@er9xuser I would like to see you test some other combinations of OpenTX firmware and receivers to see if we get any pattern. For example:
* test X series receiver with OpenTX 2.0.15 on your radio that has EU RF firmware.
* test some other D8 series receiver again with OpenTX 2.0.15 on your radio that has EU RF firmware.

er9xuser, 2015-01-20:
Ok. My friend sended me 3rd version for Taranis Plus dated 1028. This is v2.0.9.2. But he tested with the same Rx like I have and it is working.

@projectkk2glider could someone compile me opentx for taranis plus v2.0.13 with sq5fonts and lua so I can test this version if works?

I'll test X8R with 2.0.15...FYI: on X8R which was included in the box, is a sticker with 2015 date so I think that this rx has EU fw too.

And I'll try to get new D4R-II which isn't flashed with 27ms fw.

er9xuser, 2015-01-20:
FYI.

Today I've flashed Internal XJT with NON EU fw - (yeah I know it's illegal) over Mike's Maintenance mode (very good and easy to use flashing method). Then loaded opentx 2.0.15 and restored models and settings. Binded again D4R-II Rx  in D8 mode and guess what. It works like a charm.

One side of brains talking to me: Leave non eu fw and enjoy. The other side is talking: Restore back eu fw to internal xjt and help find the problem. Tomorrow I'll get D8R-II Plus with default FW and i'll test with eu fw on xjt and with opentx 2.0.15 fw.

er9xuser, 2015-01-20:
Hey guys I've nailed what is the problem. :)

Read my previous post. So then I have flashed internal XJT back to EU fw, load opentx v2.0.15, restored models and settings. Binded D4R-II Rx again and it works like a charm.

Then I've remembered that on saturday when I saw something strange on screen GLOBAL V. 
Image 1 - Acro CF250 - this model was created on stock opentx 2.0.9 FW on the Tx with Wizard. 
Image 2 - Acro CF250 - see GLOBAL V. screen - to me looks normal because I didn't touch them at all
Image 3 - MODEL05 - this model was created on opentx-taranisplus-lua-sqt5font-en-2.0.15 FW on the Tx with Wizard
Image 4 - MODEL05 - see GLOBAL V. screen - this is not ok...I didn't touch them at all. 

Conclusion: something is wrong when creating model on Tx with wizard with 2.0.15 fw.

So I'll stay with EU firmware on my xjt.
[image]
[image]
[image]
[image]

er9xuser, 2015-01-20:
Unfortunately this was not the case :(

I've turned off model and tx. Turned on again and again doesn't work.
Then I've deleted MODEL05 from Tx turned off Tx and on again. Still the same problem. After that I've rebinded Rx again and again works.
Please see videos.

https://www.dropbox.com/sh/snebl3xxtzsshu8/AABaSw20Wt6lX0Nmg0FZWL1Fa?dl=0

And another thing. Tested the same scenario like in post before with internal XJT with EU fw and X8R with EU fw binded in D16 mode on 2.0.15 fw - doesn't work.

 internal XJT with EU fw and X8R with EU fw binded in D8 mode on 2.0.15 - doesn't work.
Then removed MODEL05 turned off tx and on again...X8R doesn't work. Then rebinded and again works. Strange.

So it is clearly something wrong with creating model (quadcopter) with wizard on 2.0.15 fw.

er9xuser, 2015-01-20:
The same thing is with opentx 2.0.13 and 2.0.14 and 2.0.9.2...I found that every time I create new model and bind rx D4R-II in D8 mode...for the first time it works, but when I change model...goes to non-working state (video is showing LED sequence). So will try tomorrow with D8-II Plus from my friend (this one has stock FW from factory).  Sorry for posting...but I want to get to the bottom with this problem. I hope you'll find a bug.

kilrah, 2015-01-21:
> Image 4 - MODEL05 - see GLOBAL V. screen - this is not ok...I didn't touch them at all.

The defaults about GVAR configuration have simply changed recently. The change was likely not carried over by FrSky into their version. So nothing to see here.

At this point I'm pretty sure something's wrong with your hardware, maybe the internal module is toast or a connection going to it is flaky.
If you can see channels move with the sticks in the channel monitor, the RF mode is set to one the receiver supports, then the receiver is bound correctly according to that mode and responds at least once, then nothing that is in OpenTX' control can cause it to just not work anymore after a reboot.

er9xuser, 2015-01-21:
Increasingly it seems to me that something is wrong with internal xjt...but I'll try tonight with D8R-II Plus and see if this combo works. 

Question is why this thing happens when I create new model on Tx? As far as I know if you have for example two models in Tx and with both using 1 Rx...one just have to bind Rx 1 time to Tx module not every time when changing model?

projectkk2glider, 2015-01-21:
For D8 protocol based receivers you only have to bind once. But for X series receivers each model has its own ID (model number) and if that ID changes you have to re-bind. But if you create a new model and edit its ID to existing number (that was already bound to receiver) it will work without re-binding.

er9xuser, 2015-01-21:
For X type I know about ID. So then it is just problem with D8 protocol. We'll see tonight. Stay tuned.
I'll test again starting with flashing internal xjt to non eu fw and do all test like yesterday.

er9xuser, 2015-01-21:
Ok guys final news. Internal XJT with new EU firmware doesn't work with D8 protocol.

1. Firmware on Taranis - original opentx v2.0.9.2 - 28102014 and Internal XJT with new EU firmware.
- binded Rx D8R-II Plus(default fw on Rx)  in D8 mode to one model - working normaly
- turned off Rx. Selected another model which was already created and is set to D8 mode - working normaly
- binded Rx D4R-II(27ms fw on Rx) with one of the already created model in D8 mode - working
- then made a new fresh model, set this model in D8 mode, turned on D8R-II Plus - not working normaly - green lite flashes like on my videos from some earlier post
- turned off tx, turned on again with D8R-II Plus - not working normaly - green lite flashes strange
- turned of this Rx, turned off Tx, turned on Tx again and this time turned on Rx D4R-II - not working normaly - green lite flashes strange
- tested X8R - works ok

2. Flashed old NON-EU internal XJT firmware and did everything like in 1 test - both D type receivers works in all curcumstances except X8R doesn't want to connect because EU firmware is flashed on X8R.

IMHO...new EU FW for XJT modules - doesn't work properly with D type receivers.

Sorry guys for few posts more but I wanted to know what is the problem. In the mean time, I'll use non eu fw on internal xjt and wait for FrSky to polish few bugs or make new EU firmware for D type receivers and this is urgent. Plus if they will make new EU fw for D type Rx's...please tell them to make 27ms EU FW for D4R-II.

And please post info from FrSky about this if you can.

Thanks!

Br,
er9xuser

projectkk2glider, 2015-01-21:
Finally some good news (at least for OpenTX). I will make sure that FrSky is aware of this issue. 

Until this is resolved by FrSky, I suggest keeping this issue open.

ckleanth, 2015-01-21:
What exactly is the change of the eu regs, frequency range or rf power? If its the latter the  choice between the old firmware (that works) and the new eu firmware that complies to some eu regs an elementary one

bsongis, 2015-01-21:
Thanks @er9xuser for your tests and reports here!

er9xuser, 2015-01-21:
No problem!

Thanks to all developers and contributors!

kilrah, 2015-01-22:
I think it's worth doing our own test before reporting to frsky. Again I have read several reports of people testing D8 on the new FW and having no problems.

I'll have a go myself when I find time.

er9xuser, 2015-01-22:
My question would be...why everything works normaly with non eu fw but on eu fw fails. But I see you've edited your post. Is this answer from Mike on openrcforums?

kilrah, 2015-01-22:
Well the most likely explanation would be that the EU firmware has a bug. We'll see what they say.

projectkk2glider, 2015-01-22:
FrSky has just notified us that they have identified and **fixed this issue**. The new firmware will be available on their site soon. When the tests of new firmware prove successful, we will close this issue.

projectkk2glider, 2015-01-23:
FrSky was very responsive, the updated firmware is available for download at http://www.frsky-rc.com/download/view.php?sort=&down=191&file=XJT%20Firmware-%20EU%20v1.8.1-%20Bug%20fixed All X series receivers with EU firmware  are also affected by this and have their firmware also updated on FrSky site.

Mike has already tested it and confirmed that it works. I will close this. @er9xuser test and report back.

closed 2015-01-23 by projectkk2glider

er9xuser, 2015-01-23:
Great! But why they didn't remove old buggy firmware from their site? And I didn't found new FW for X series Rx.

I'll test and report back.

LapinFou, 2015-01-23:
X4R-X4RSB: [url]http://www.frsky-rc.com/download/view.php?sort=&down=187&file=X4R/X4RSB-%20EU-%20SBUS%20Bug%20Fixed[/url]

X8R-X6R: [url]http://www.frsky-rc.com/download/view.php?sort=&down=188&file=X8R/%20X6R-%20EU-%20SBUS%20Bug%20Fixed[/url]

XJT (D8 mode fix): [url]http://www.frsky-rc.com/download/view.php?sort=&down=191&file=XJT%20Firmware-%20EU%20v1.8.1-%20Bug%20fixed[/url]

er9xuser, 2015-01-23:
I found fixed fw for XJT. But at first didn't saw new fixed fw for X type receivers...thanks. I've edited my previous post.

LapinFou, 2015-01-23:
You're welcome.

er9xuser, 2015-01-23:
FYI

Flashed again NEW XJT EU fw on Taranis Plus. Tested everything and confirmed working.