opentx/opentx #2032: CCPM extension to have weights instead of reversals

ArchmageAU, 2015-01-20:
Currently in the heli menu, OpenTX only allows you to reverse input channels that are inputs into the CCPM mix.  This is a request to change those from check box reversals into full weight numbers.

When programming a CCPM heli, a user can have multiple pitch curves and different DR's for aileron and elevator.  At the moment for CCPM program, these need to be re-adjusted to give input weights to the CCPM algorithm (which can be done by using global variables, but this is very clunky).

Can these reversals just be input weights percentages from -100.0 to +100.0?  All the weights do is multiply each input by the weight percentage before processing by the CCPM algorithm.  (Ail by Ail%, Ele by Ele%, Collective by Collective%).

Existing models could be translated to have weights of -100% or +100% (based on current check boxes).

For more information see open discussion on HeliFreak: http://www.helifreak.com/showthread.php?t=676944

kilrah, 2015-01-20:
Why doesn't setting weight in the input work for you?

I would think the opposite, the invert setting in the heli page is obsolete and could be removed (pre-2.0 D/Rs were positive only, hence why it was required).

kilrah, 2015-01-20:
That said, I'm still for a choice of which input to use for lateral and longitudinal cyclic since 2.0 was in development to avoid the "obscure" internal mapping a user can't really know how it's done.

projectkk2glider, 2015-01-20:
When one uses negative weight in Inputs, the trim for that channel does not invert and now behaves in opposite direction - ie trim up actually moves servo in down direction. So you have to take that in consideration if you choose to use negative input weight.

kilrah, 2015-01-20:
Ah yes, forgot that. Inverts need to stay there then, but I still don't see a need for additional weights.

Still voting for a selection of the actual input though, like for collective.

ArchmageAU, 2015-01-20:
Reversals are inconvenient when you have multiple pitch curves (like multiple flight modes).  If I have a normal mode, an idle up 1, idle up 2, idle up 3, and hold mode, there are 5 weights I need to configure for inputs (unless I do them in the mixes page with replacements for each then finally do a multiply mix at the end).

If you have 3 cyclic DRs, 3D, Normal and F3C. you need to alter each of these weights separately (as the weight must be applied to the last [I]Input derived from Ail and Ele).  So your cyclic weights are DR multiplied by CCPM weight.  You cannot do additive mixes in the input screen.

With the advent of configurable inputs.  The best place to put the pitch curves are as a customizable input (not a series of replace mixes).  If using this, like the cyclic weights, you need to adjust your curves to output the re-weighted collective as the input to the CCPM.

By adding the weights directly into CCPM, all the input and DR calculations occur after the logical mappings for control are done.

You can get around all this, but I thought the idea is to make life easier.

My understanding of the controls are map the controls to a pure model (Ail, Ele, Thr, Rud, Gyr, Col all -100/+100), This pure model is the output of the "Inputs".  Next is to map the interactions between pure controls with "Mixes" to get pure model channel controls.  Then finally map the pure model channel controls to the physical channels (at this stage CCPM should be introduced).  At the moment CCPM is done between pure model "Inputs" and pure model channel controls "Mixes".

CCPM should also have the ability for phasing (as cyclic phasing may not be exactly 90 degrees and unable to be fixed mechanically, but that is another story).

ArchmageAU, 2015-01-20:
As for negative weights and trims.  Trim is for the "Input" direction (ie steer left, push forward).  Negative weights are to translate from the "Input" to the CYC mix.

Negative weights in CCPM mix will not have an adverse effect on trim (as long as the OpenTX user programs their model correctly - ie. for inputs, +Ail = Right, +Ele = Nose down, +Col = positive pitch which is the same directions as the trims).

kilrah, 2015-01-20:
> Reversals are inconvenient when you have multiple pitch curves (like multiple flight modes). If I have a normal mode, an idle up 1, idle up 2, idle up 3, and hold mode, there are 5 weights I need to [...]

Well you'll have 5 curves on the pitch inputs (or more likely 4 curves with a weight of 100% and an "xx% MAX" line for hold). Weight on the heli page won't change a thing there.

> If you have 3 cyclic DRs, 3D, Normal and F3C. you need to alter each of these weights separately 

Sounds normal, the whole point of having 3 DR entries is precisely to have different weights...

> (as the weight must be applied to the last [I]Input derived from Ail and Ele)

It somehow sounds like you're misunderstanding inputs, not sure how though.

Or you're somehow trying to get your cyclic weights to be a combination of FM AND DR... but that would seem twisted to me, seems to me those are typically separate. One group of pitch modes, one group of cyclic modes that are independent. 

A sample setup would look like that:

[image]

[image]

ArchmageAU, 2015-01-20:
I think the misunderstanding is how CCPM works.

I set up a model to demonstrate.

[image]
[image]
[image]
[image]

Note: This is CCPM 120.  When SG is down, then cyclic DR is 100 (expo 20).

When I am at 50% cyclic (just under because of expo), the max collective I can have is 59% before CH2 hits it's limit.

[image]

If I push to 100% collective, CH2 cannot go past max but CH3 and CH6 continue to move.

I have cyclic ring of 100, so without weights, I can push servos to bind points.  In fact 100% back elevator puts CH2 to 100%.

CCPM weights enable the cyclic controls to remain the expressed DR values without having to adjusts.

If a heli has a desired collective pitch of 12 degrees and a desired cyclic pitch of 8 degrees, the cyclics should be reduced to 40% and collective should be weighted to max at 60%.  This way CYC1, CYC2 and CYC3 will all vary from -100 to +100 as desired (reflecting the controls).

Normal pitch curve would need to change from: (-40,-100 / 0,0 / +100,+100)
[image]
to: (-24,-100 / 0,0 / +60,+100)
[image]

As stated before weighting in CCPM allows pure model configuration, rather than weaving the physical model constraints into the pure model construction.

ArchmageAU, 2015-01-20:
kilrah,

In your example.  If SC is UP (100% elevator DR).  How much positive collective can you add if you pull back on the elevator before CH1 maxes out?

projectkk2glider, 2015-01-20:
@ArchmageAU I have zero experience with helis, but if all you want is to prevent binding of cyclic channels, you must appropriately set the **Swash Ring** value. The value in your setup is 100, meaning that full throws are possible. Try reducing that number. 

Or maybe I misunderstood and you need another limiter that takes in account the collective and the pitch together to limit the cyclic servo movements? Huh, this heli stuff is so confusing...

ArchmageAU, 2015-01-20:
@projectkk2glider, I have a little more RC heli experience (main author for the heli articles on OpenTXU, admin on HeliFreak, etc...).

Swash ring limits the input aileron and elevator stick inputs so stick inputs from Ail and Ele stop at the swash ring limit.  ie. swash ring 50 means if the movement of Ail + Ele move more than 50% away from physical center stick, they will not go any further.

What is needed is not a limiter, but an input weighting so that the combination of these 3 inputs (Ail, Ele, Col) for an RC heli can give non-clipped results (for CYC1, CYC2 and CYC3).

Unfortunately there is no equivalent for planes, gliders or multi-rotors.  So unless you are experienced with CCPM heli setup, this will be confusing.

If you were to manually write the mix formulas for CYC1, CYC2 and CYC3, they would be:
CYC1 = (([I]Ail * 0) * ARev) + (([I]Ele * -1) * ERev) + (([I]Col * 1) * CRev)
CYC2 = (([I]Ail * 0.865) * ARev) + (([I]Ele * +0.5) * ERev) + (([I]Col * 1) * CRev)
CYC3 = (([I]Ail * 0.865) * ARev) + (([I]Ele * +0.5) * ERev) + (([I]Col * 1) * CRev)

note: ARev, ERev and CRev are 1 or -1 dependent on reverse status.

Without being able to change the weights of the inputs, these equations clip at 100 quite readily.

Till now the solution has been to adjust the [I]Input weights, but as more are added, this become cumbersome and confusing to maintain.  The ideal solution is to add weights to the mix equation.  ie. allow ARev, ERev and CRev vary from 1 to -1 in 0.001 increments and represent like weights as -100.0 to +100.0

By allowing ARev, ERev and CRev to vary, the ratio between collective pitch and cyclic pitch can be represented.

For most flybarless helis, this is not an issue (as CCPM mixing is done by an onboard controller).  For flybarred helis (still used for F3C competition and preferred by some modelers), transmitter CCPM mixing is important.

kilrah, 2015-01-21:
The simple, obvious solution (which is exactly the same for a fixed wing model by the way!) is to set your DRs to the values needed to avoid binding.

You will never have an "Ail" or "Ele" DR at 100%. Maybe you'll need to have the weights at 53% for your full rates so that it doesn't clip at any pitch and cyclic combination. Lower rates will be maybe 60% of that, so 32%.

I know we naturally like seeing a round number and even better a 100% for full rates - but if you forget that "artificial limit" there should not be a problem. Of course if you want to be able to have the absolute maximum possible control authority for all different pitch curves it will take a lot of lines, but it will work (and I don't believe it's a common thing to do in the first place). 

As I said it's the same for fixed wing. With non-mixed controls you can have 100% input weights but you'll have clipping as soon as your trim is not 0. With mixed controls you just have to have lower input weights. So as soon as you care about your model setup a bit (I have foamies with everything at 100% and don't give a damn about imprecisions, but on my good gliders I'll go all the way).
It's just the way to deal with it on OpenTX. It's different from some others, but not limiting when you're aware of it.

kilrah, 2015-01-21:
Post above edited, don't look at the email notification.

ArchmageAU, 2015-01-21:
Core is: is it worth implementing weights for CCPM instead of just reversals or not?

Sure you can program around it.  It's complex and clunky to do so, but it can be done (and not real easy to do at the field either as there is math to contend with).

In my mind, ideally you shouldn't have to.

- Spektrum CCPM you set mix values.
- Futaba CCPM you set weights (AFR).
- (Have not looked into JR and Jeti, but pretty sure they use weight type mechanisms too).
- OpenTX you currently program around the absence of weights for CCPM.

Yes, OpenTX is different, but it should also be better (and if we really want to add value put in phasing as well, but that is more complicated).

I do not want to instruct people on OpenTXU to keep in mind their fixed CCPM weights and ratios each time they want to re-visit a models programming (and have to enter it into every pitch curve and DR setting individually).  To my mind this does not make sense (and even harder to explain it).

@kilrah, are you a CCPM heli pilot?  Do you have to deal this side of the heli TX programming regularly?  This is the one part of TX programming that is vastly different from anything you do in planes or gliders.

ArchmageAU, 2015-01-21:
Summary of what I am asking for is:

- update SwashRingData in myeeprom.h to add weights.
PACK(typedef struct t_SwashRingData { // Swash Ring data
  uint8_t   invertELE:1;
  uint8_t   invertAIL:1;
  uint8_t   invertCOL:1;
  uint8_t   type:5;  
  uint8_t   collectiveSource;
  uint8_t   value;
  int16_t   weightELE;
  int16_t   weightAIL;
  int16_t   weightCOL;
}) SwashRingData;

- Change mixer function in mixer.cpp
from:
    if (g_model.swashR.invertELE) vp = -vp;
    if (g_model.swashR.invertAIL) vr = -vr;
    if (g_model.swashR.invertCOL) vc = -vc;
to:
    if (g_model.swashR.weightELE) { vp = (vp * 1024) / g_model.swashR.weightELE; } else { vp = 0; }
    if (g_model.swashR.weightAIL) { vr = (vr * 1024) / g_model.swashR.weightAIL; } else { vr = 0; }
    if (g_model.swashR.weightCOL) { vc = (vc * 1024) / g_model.swashR.weightCOL; } else { vc = 0; }

- make an update function to convert models with old style to new style on model load.
    eg. during conversion.
    g_model.swashR.weightELE = g_model.swashR.invertELE ? -1024 : 1024;
    g_model.swashR.weightAIL = g_model.swashR.invertAIL ? -1024 : 1024;
    g_model.swashR.weightCOL = g_model.swashR.invertCOL ? -1024 : 1024;

- Update opentx heli input/show to store weights in weightELE, weightAil and weightCOL.

- Update companion heli screen to input/show weights from weightELE, weightAil and weightCOL.

I do not have the time/setup to join the programming effort myself.

kilrah, 2015-01-21:
> Futaba CCPM you set weights (AFR).

Which is exactly the same thing you do on OpenTX. The Ail AFR you set on Futaba is exactly the same as the Ail weight you'll set in OpenTX' inputs page. 

> OpenTX you currently program around the absence of weights for CCPM.

Well that's how you see it, but to me it is setting it where you should be setting it in the first place. You want to adjust how much travel the cyclic has - nothing more natural than to set the cyclic's rate, is there?

> are you a CCPM heli pilot? Do you have to deal this side of the heli TX programming regularly? 
Not regularly, but I have set up and flown CCPM helis both on Futaba and OpenTX. 

Futaba offers you CCPM mix weight adjustment, as they say, to compensate for mechanical imprecisions. On OpenTX those are compensated for using the SERVOS page. Which IMO makes more sense, as you're acting on individual components that are easy to manage instead of multiple if you do it before the mixing.

483917, 2015-01-21:
@ArchmageAU ,  I agree completely, it is far too complicated to make a correct CCPM setup with OpenTx, this is an important feature for CCPM helicopters that should be included in OpenTx as standard without a doubt.
OpenTx / FrSky is a wonderful system, I can not imagine using anything else but the helicopter part needs to be improved.
I would gladly do the job but i lack the programming experience/skills needed

projectkk2glider, 2015-01-21:
Maybe we have a different kind of problem here. Currently the input selection for Elevator and Aileron for cyclic mixing is done auto-magically. That means that OpenTX auto-discovers which input it should use for the elevator and aileron values. I was not avare of this till now and I dont like this *feature*. The auto-discovery currently works like this (example for elevator):  for elevator value use the last active line from inputs that has the source set to Ele. 

This auto-discovery can lead to all kind of undesired side-effects. For example if one would define additional lets say input 7 with Ele source that line would be used for cyclic calculations. 

Shouldn't there be an additional settings for heli where one could choose which source to use for heli aileron and elevator (same as collective source which can already be selected)? Then one could even use a mixer output for heli elevator input. Maybe this additional flexibility could be used as a cyclic weight like OP is trying to suggest.

kilrah, 2015-01-21:
> Shouldn't there be an additional settings for heli where one could choose which source to use for heli aileron and elevator (same as collective source which can already be selected)? 

Yes! That's what I've been saying for more than a year, and again at the top of this thread :)

ArchmageAU, 2015-01-21:
Would be happy with that even.  This way ALL the inputs could be mapped to virtual channels which have the weighting which then has the CCPM mix applied to output CYC1, 2 and 3.  (easy to program and understand).

One thing in OpenTX parsing I do not know.  Are the mixes processed in numeric order?  If so, would it take 2 parse runs to resolve changes in input.  (1st run inputs and mapping to high channels. 2nd run mix against high channel and map to CYC1, 2 and 3 for output on low channels).  How many times per second are parse runs done (and what impact does this have on latency)?  Latency is an issue for heli 3D (timing cyclic with tail), but not so much for F3C.

kilrah, 2015-01-21:
> Want an example? Set up a CCPM heli that does not bind with 3 DR's and 3 Pitch curves with a collective/cyclic ratio of 10:8. Then change the same model to have a collective/cyclic ratio of 13:9 (like you copied it to template another model).

You'll have to change 6 weights. On top of changing every value you have on the servos page, for which you have to basically either wipe the inputs or add temporary lines anyway. Seems negligible to me...

> One thing in OpenTX parsing I do not know. Are the mixes processed in numeric order? 

Yes

> How many times per second are parse runs done (and what impact does this have on latency)?

The mixer detects if there are nested values, and if yes recalculates everything right behind. With a mixer time that rarely exceeds 2ms and an output that only has a 9ms cycle time, it's unnoticeable.

ArchmageAU, 2015-01-21:
Thanks for the clarification on the parsing.

@kilrah, I must really be missing the point.  How is altering 6 values (well 7 actually) in my example, better than altering 3?  How is having to compute for cyclics: x% / (8/18) * (9/22) for each weight adjustment easy?  (At least the single weight on the collective on a phantom channel changes from 56% to 59%)

Yes, you CAN do it (changing 7 weights - 6 in Inputs, 1 in Mixes).  But is this the way we SHOULD do it?

Changing all the appropriate weights outside the CCPM mix is just not easy (and if you get it wrong it's hard to tell).

Granted, you would need to change the servo PPM center and weights.  You are changing model after all.  But you set these to the servo limits by manual testing (ie. standard physical model mapping, 0 pitch testing, max/min collective testing).

Being able to select the Ail and Ele inputs for the mix would be a good start.  But as far as I can see, the IDEAL solution is to do like what all other TX manufacturers have done and program weights into the CCPM mix (ie the Heli page/panel).

Do we want OpenTX adequate or best?

kilrah, 2015-01-21:
>  How is altering 6 values (well 7 actually) in my example, better than altering 3?

It's of course not, I never said that. My point is, is altering 7 vs 3 so bad that it's worth spending a few days in development time changing it when there are close to 200 pending issues, of which many are bugs, things that impact most users, or enabling things that are not possible at all? How often do you have to go through that on a heli setup? Once/twice?

> How is having to compute for cyclics: x% / (8/18) * (9/22) for each weight adjustment easy? (At least the single weight on the collective on a phantom channel changes from 56% to 59%)

My problem is that I don't understand why to do such complex calculations that in the first place... - I'd adjust the pitch curves to get the +/- degrees needed (checking with pitch gauge) if required, then check cyclic for no binding / adjust for max throw (or desired throw) as desired and call it a day. If heli is too nervous reduce rates by 5-10% at a time and there we go. 
But then I'm a "practical" guy, not a theorist. I move the stick, watch that the thing does what it should do, if not adjust until it does, and the value will be what it ends up being. If another value has to be half that, then I halve that.  

> Do we want OpenTX adequate or best?

As good as possible with the available resources. And...

> I do not have the time/setup to join the programming effort myself.

as that's unfortunately what everybody says we have to choose and prioritize, and "comfort" features tend to fall to the bottom of the pile as we'd rather first fix bugs, then give people a way to do things they can't do at all, then fix confusing things that cause repeated support questions before improving things that can already be done in a satisfactory-but-just-less-than-ideal way.

This issue will of course stay open for whenever someone wants to have a go at it. At this point I'd put the cyclic source selection in both the "allow to do things that can't be done at all" and "confusing things" categories, and the weight in the "comfort" section.
Fixing the source selection will already improve your case, and who knows while in the subject cyclic weights could be added at the same time.

ArchmageAU, 2015-01-21:
> This issue will of course stay open for whenever someone wants to have a go at it. At this point I'd put the cyclic source selection in both the "allow to do things that can't be done at all" and "confusing things" categories, and the weight in the "comfort" section.
Fixing the source selection will already improve your case, and who knows while in the subject cyclic weights could be added at the same time.

This is what I ask.  Put as an enhancement and tentative schedule to 2.1 or 2.2.  It's not a this must be fixed tomorrow.

It's not a bug, as it can be worked around.

I had taken most of the debate as to whether this would be closed (which is what I wanted to avoid).  I believe it is a worthwhile change, I just want it in the pipeline, somewhere.

As any resolution to this will involve a change to the eepe format (extension of the SwashRingData), it is best scheduled for an update that changes the eepe.  (like 2.0 did from 1.x).

I'll give some time to try set up a development environment on a machine.  It will take time to get up to speed with the development methods being used.  It just may take a while.

> > Do we want OpenTX adequate or best?

> As good as possible with the available resources. And...

Could ask nothing more.

I am a staunch supporter / advocate of OpenTX.  Just currently lack the committed time / environment to also be a developer.

jermoo, 2015-02-03:
This is like a holy grail for ccpm heli pilots. This and having the ability to script curves would push Open Tx out to more pilots as it would be easier for us to have non-technical people able to program their models via a lua. Please implement this even if it's just 3 settings on the CCPM menu.
Thank you very much for all that you do.
Jer

bsongis, 2015-02-05:
I am having a look at this one. First I will add a configuration for ELE and AIL as it is *really* needed. Then I am ok to add weight (-100% => +100%) instead of INVERT on all 3 inputs, is that what you need?

Would you take care of the tests? I understand about nothing to helis and CCPM.

ArchmageAU, 2015-02-06:
Yep, -100 to +100 on the 3 weight is exactly what is needed.

The ultimate is if the source for Ail, Ele inputs could also be specified.  (at the moment, these are interpreted by the last Input that uses a curve derived from the appropriate stick, but this can be fooled, and is no good for buddy boxing).  I only mention this as it is done is exactly the same place in the code as where the weights would be implemented.

Yes I am happy to test this.
(and will get the http://open-txu.org/ccpm-for-taranis/ article amended - and the HeliFreak one too: http://www.helifreak.com/showthread.php?t=678601 )

commit 1e76e24d (2015-02-15): Merge pull request #2057 from opentx/bsongis/issue_2032_CCPM_on_taranis

Fixes #2032
commit 4a9c87cc (2015-02-15): Fixes #2032
closed 2015-02-15 by bsongis