opentx/opentx #2040: Problems with setting Expo from gvars in OpenTX / OpenTX Companion

StewLG, 2015-01-24:
I've been getting some strange-to-me, possibly buggy behavior involving Expo on GVar 3 and Curve 4. It seems like the emulator/actual firmware is interpreting the setting of gvar controlled expo in the OpenTX Companion interface to be something involving curves, which if true makes some sense since I can't find any setting for gvar controlled expo in the emulator or real firmware, only curve control.

Here's a case I'm looking at right now:

[image]

So, there you can see I've requested Expo be controlled by GV3 for both channels 2 & 3, yet the simulator is interpreting this to mean that this is an instruction about Curve 4.

Here's what that looks like once I set Expo back to Diff 0 for both channels:

[image]

The mention of Curve 4 is gone.

More background available here:

http://openrcforums.com/forum/viewtopic.php?f=45&t=6622

ArchmageAU, 2015-01-27:
Can you use GV for weight and expo in the inputs screen instead of the mix screen?

(Worked when I tried on my Companion 2.0.15 for 9XR-Pro firmware with gvar enabled).

Do you need to use expo on mixes, or would inputs be enough.  When working through the firmware, looks like expo on mixes for 9XR-Pro (normal or gvar) is not supported.

StewLG, 2015-01-27:
[image]

Thank you for this suggestion - I think it will work fine for me to do gvar controlled expo from the inputs section for my particular use case. Anyone else who has this problem in the meantime can benefit from this idea; see the screenshot I just included.

That said, I believe there is still a bug in the firmware/simulator/OpenTX Companion, so I will leave this issue open. Either you should be able to set gvar controlled expo from the mixes and have it work properly, or you should be prevented from setting it. Right now there is a disjunction between OpenTX Companion and what the firmware can actually do.

ckleanth, 2015-01-27:
Everything works just fine....
(I think this is my model from r2940 so you might need to setup the edge function again)

http://rcsettings.com/index.php/viewdownload/3-fixed-wing/68-fixed-wing-5-channel-with-flaperon-implementation-on-flight-modes-two-modes-gyro-setup-with-variable-master-gain-throttle-hold-expo-dual-rates-with-custom-curve-linear-part-is-the-same-for-all-rates-the-exponential-part-has-high-medium-low

StewLG, 2015-01-27:
CKLeanth, the setup you link to, is that for Taranis? (The site says "OpenTX Model templates for FrSky Taranis".) If so, I think you'll find it different for the 9XR-Pro, whose implementation of OpenTX is less complete. I expect this bug is specific to the 9XR-Pro.

bsongis, 2015-01-27:
Expos are not available in Mixes on 9XR-PRO, I should have disabled them in Companion / 9XR-PRO as well.

closed 2015-01-27 by bsongis

StewLG, 2015-01-28:
Thanks as always Bertrand. At least it will not be confusing any more.

commit 116fe0dc (2015-01-27): Fixes #2040
commit d77bfae7 (2015-01-27): Fixes #2040