opentx/opentx #2042: Helicopter CCPM120,CPPM120X and CPPM140 are not right ?

483917, 2015-01-25:
WindowsXP SP3 swedish
Companion 2.0.15 swedish
Taranis B , OpenTx 2.0.15 sqt5font, Haptic, LUA ,swedish

To day i setup my new KDS450SV helicopter "Trex450 clone" it has CCPM120 swash,
and noticed that the elevator Cyc2,3 servos had only half of the elevator Cyc1 servo deflection ?
But when I changed to CCPM140 I got equal deflection on all 3 servos while moving elevator stick

so i looked in "mixer.cpp from row 621 and down" and it does not look right ?? or am i wrong ??

To reproduce, create basic helicopter with swash 120 and in mixer set Cyc1,2,3 weight to 50%
start simulator and set pitch stick to 0% and move elevator stick up/down and look at Cyc1,2,3 values
then change swash to 140 and do the same test

ArchmageAU, 2015-01-25:
Agreed for CCPM 140.

From mixer.cpp

#define REZ_SWASH_Y(x)  ((x))   //  1024 => 1024
...
      case SWASH_TYPE_140:
        vp = REZ_SWASH_Y(vp);
        vr = REZ_SWASH_Y(vr);
        cyc_anas[0] = vc - vp;
        cyc_anas[1] = vc + vp + vr;
        cyc_anas[2] = vc + vp - vr;
        break;

(FYI: vc = collective, vp = elevator (pitch), vr = aileron (roll) - values are -1024 to 1024)

CCPM 120 is correct as for elevator, cyc2, cyc3 should move half the distance as cyc1 (and in the opposite direction).  CCPM 120X is also correct (as this is CCPM 120 rotated 90 degrees).

The quick way to fix 140 CCPM is:
#define REZ_SWASH_W(x)  ((x) - (x)/4 - (x)/16 - (x)/32 - (x)/128 - (x)/256 - (x)/512)   //  1024*sin(140) ~= 658
#define REZ_SWASH_V(x)  ((x) - (x)/8 - (x)/16 - (x)/32 - (x)/64)   //  1024*cos(140) ~= 784 
...
      case SWASH_TYPE_140:
        cyc_anas[0] = vc - vp;
        vr = REZ_SWASH_V(vr);
        vp = REZ_SWASH_W(vp);
        cyc_anas[1] = vc + vp + vr;
        cyc_anas[2] = vc + vp - vr;
        break;

Ideally, (if possible), a degrees for CCPM and a phasing should be able to be entered.  120X would be 120 with 90 phasing, and any degree combination from 90 to 180 should be permissible.

The math becomes:
        cyc_anas[0] = vc - cos(phasing)*vp + sin(phasing)*vr;
        cyc_anas[1] = vc - cos(phasing+degrees)*vp + sin(phasing+degrees)*vr;
        cyc_anas[2] = vc - cos(phasing-degrees)*vp + sin(phasing-degrees)*vr;

The tricky bit is efficiently working out the cos and sin values.  In the current implementation, these are hard coded.

483917, 2015-01-25:
Ok, today i deleted my setup from yesterday and made a new one,
CCPM120 and swash worked as it should do ?? so you are right CCPM120 is working properly, what happened in my first setup I have no idea? stupid user error maybe ?? :-)
Shall this issue be closed and maybe create a new for CCPM140 ?

ArchmageAU, 2015-01-25:
Just change to CCPM 140 not correct (as that is currently wrong).

483917, 2015-01-25:
Yes ,thanks

flyandhike, 2015-02-09:
CCPM120 correct??

I try to program my T-Rex 550 FBL, so far it works.
But: when I move the Roll-Stick completly form left to right on a far end of the Nick-Stick, I recognise a nick movement on the swash-plate. Nevertheless, the Nick-Stick will not be touched (as it will be left allways at the far end), so never a Nick-moment should arise, even, if this occurs on the far end edges only (outside of the green circle).
It will be easy to test, also on the OpenTX Companion Simulator (Ch1=Cyc1; Ch2=Cyc2; Ch6=Cyc3):

[image]

I assume, this is a result out of the trigonometric function in the CCPM120 calculation. I'm not familiar about the calculation program, but be aware, that the trigonometric function change the sign, when it comes outside of the green circle.

Of course, a missinterpretation or bad programming on myside is also feasible.

Best regards
Donat

jermoo, 2015-02-09:
I didn't think we used CCPM in the radio for FBL controllers?

flyandhike, 2015-02-09:
Some they use, some don't. The 3GX from Align does use the CCPM as input.

ArchmageAU, 2015-02-13:
@flyandhike Have a look at http://open-txu.org/ccpm-for-taranis/  It goes through CCPM120 in detail as to how to set up and what to expect.

CCPM 120 works fine. (unless you can detail a case where it does not).

At 0 elevator, moving aileron only should not change the cyc1 output value.  At positive or negative elevator (like your example), if you have swash ring on, then moving aileron can move elevator if you move outside the swash ring (keep angle from center at swash ring distance).

flyandhike, 2015-02-16:
@ArchmageAU : yes, I have allready read this document, where I struggled through the ring function. By setting it to 100 (as recommended), the movement of the elevator is obeserved, when moving outside of the ring with the stick. This is in accordance, what you describe and I recognised. I don't know why it is like this, but this seems for me not to be correct, as I will not be able to control the stick movement only inside of the invisible ring.
Nevertheless, so far, I do not need the ring function and I have set it to 0, then it works fine.
By the way, the value 0 is not logic for 'ring off', as 100 (%) is the full range without limitation and the lower the number, the smaller the cylindric ring. So, 100 means for me 'ring off', and 0 means full limitation = no CYC at all. It's a matter of opinion.

closed 2019-11-15 by stale[bot]