opentx/opentx #2043: Add way to start LUA Telemetry directly after power up

fzahn, 2015-01-25:
there is no way to autostart a lua-telemetry-screen after power up. Currently you have to enter the telemetry-screen by long-holding the Page-Button. It would be cool, if there would be a option in each model-setup to tell the Taranis to directly enter the telemetry-screen after power-up

mhotar, 2015-01-25:
I would prefer to unify all the front pages we have with telemetry ones into one queue and simply make it to remember last used page in selected model. Then we will be able to use Page/LongPage to cycle through them in both directions. With Lua function like drawMainPanel, which will draw top bar and trims, it would be nice to do cool custom pages.

bsongis, 2015-01-25:
Right this is something I have done in my workspace (one list for all those screens). It will be good to have it on Taranis as well.

AndrewFarley, 2017-10-04:
+1 +1 +1.   This feature can't be that hard to implement... I'm bumping it, can we get this implemented?

lshems, 2020-04-08:
not stale, but perhaps covered in another issue?

AndrewFarley, 2020-10-08:
+1 on not being stale, still desired

closed 2021-09-04 by stale[bot]

AndrewFarley, 2021-09-08:
+1 on not being stale.  Still relevant.  But clearly no one cares enough to implement it.  :P