opentx/opentx #2047: Extention of multipos switch

flarssen, 2015-01-28:
It would be nice to have the multipos switch extended beyond six positions, preferably to nine (Sx1-Sx9). This would make it possible to use momentary switch/resistor arrangements for direct selections of flight modes etc. One "position" will be no switches pressed, so this would give support for 8-way switches.

matbogdan79, 2015-02-12:
Wold be better to have switches programable so we can use them for other sliders for example mooded taranis with joistick for camera gimball.

flarssen, 2015-02-13:
Not sure what you mean. 
What I have in mind is just a change in existing functionality; change multipos switch from 3/6 positions to 3/6/9 positions, although 10 would support 9-way switches like this http://www.ebay.com/itm/NAVIGATION-SWITCH-9WAY-RKJXM1015001-1435771-/271661046719?pt=LH_DefaultDomain_3&hash=item3f40429bbf
For a gimbal, this would let you change directions in steps, and return to centre by pushing down. 
But this I just one way of using this extension. OpenTx has 9 flight modes that could be selected easily, etc.

matbogdan79, 2015-02-13:
I totaly agree with you. Bu you are missing some point here.
A 3 way switch is a 3 contactor meaning 1 5v wite + wires that go in digital/analog ports of the microcontroler. If some of the switches go to analog port than we can have 9,12, 15, infinite position switch. If they go to only digital in ports than we are stuck.
Don't confuse microcontroler input with servo output. When you move a 3 way swtitch practicaly you put 5V to each of the digital in pins.
Having one of them as analog in, you can mount, for example in your case, 9 different resistors for each of the positions ( 1k, 1.2k, 1.5k, 1.8k, etc.) than with the correct firmware you can map the each input value to a function ( example flight modes ).
This is why i asked if switches can have potentiometers (Joistick).

Another way is to use the 3 way switch with an left/right encoder that detect if you moved to the right or to the left the knob and than pulse to pos 1 or pos 3 of the switch and not touching the knob mean pos 2.
This is also possible with the right firmware modification.

I am planing to mod my radio with a encoder knob to replace my +/- buttons for easy navigation and settup. This is the knob ( http://www.ebay.com/itm/12mm-Rotary-Encoder-Switch-Flat-Top-Knob-NEW-/201285173915?pt=LH_DefaultDomain_0&hash=item2edd88029b ) or something like this one.

flarssen, 2015-02-13:
Quote:
"Having one of them as analog in, you can mount, for example in your case, 9 different resistors for each of the positions ( 1k, 1.2k, 1.5k, 1.8k, etc.) than with the correct firmware you can map the each input value to a function ( example flight modes )."

That's what multi position switch is all about. It replaces (analog) S1 (S2) and when calibrated, firmware provides virtual switches S11-S16 (S21-S26). I would like S11-S19 (S21-S29) er even better, S10-S19 (S20-S29)

closed 2019-11-15 by stale[bot]