opentx/opentx #2049: Missing download for 2.0.15/Mac

dustin, 2015-01-30:
There's a 2.0.15 release link here:  http://www.open-tx.org/2015/01/13/opentx-2.0.15/

It points to http://downloads-20.open-tx.org/companion/companion-macosx-2.0.15.dmg which is non-existent.  The latest version there seems to be 2.0.13.

projectkk2glider, 2015-02-04:
Unfortunately we don't have automatic build solution (yet) for MAC version of Companion. Therefore each release must be compiled *by hand* and the 2.0.15 for MAC was not done yet. Related issues #1482.

dustin, 2015-02-06:
That's unfortunate.  :(

Is there someone who is planning to roll a 2.0.15 release?

andyjenkinson, 2015-02-20:
I have done another ad-hoc release and updated the website. The previous links should now work.

closed 2015-02-20 by andyjenkinson

projectkk2glider, 2015-02-21:
@andyjenkinson  thanks!