opentx/opentx #2050: Less memory available for Lua in 2.0.15

projectkk2glider, 2015-01-30:
People on forums report that some Lua scripts won't run anymore on 2.0.15 and that the reason is less memory. We need to check this. 

On the other hand it might be something else in script that is wrong.

bsongis, 2015-01-30:
NANO is available

on4mj, 2015-03-18:
I'm one of those people :-)
My <a href="http://rcsettings.com/index.php/viewdownload/13-lua-scripts/179-f3k-training">F3K training script</a> stopped working on the TX when 2.0.15 was released (but it would still work fine in Companion).

I didn't keep the notes I made when I solved the problem, so I don't have exact figures anymore.
What I can tell is that I used the new debugging feature in Companion to get memory usage data from the GC.  Sorry about that.

I first reduced the memory usage in my previous version by about 400 bytes.  The script would start again, but crashed along the way as more memory was allocated.

In the end, I gained about 1.000 bytes with diverse space optimizations (mostly removing 'closures') and that definitely solved the problem.  I could even reintroduce a couple of new features.

In the process, it seems that I also hit an arbitrary limit put on the variables amount (number of vars and functions global to the script, I guessed).

That those errors are reported as "Lua syntax error" doesn't help to identify the problem :-/

bsongis, 2015-03-19:
OK. I am afraid it's the NANO=YES option that we introduced in 2.0.15 which makes the difference. Is it the same in the nightly builds (stable ones, 2.0)? If yes, I will try to go back to NANO=NO to see if it fixes your problem. I wait for your test on nightly build then.

on4mj, 2015-03-21:
Did it.  Unfortunately, I can't draw any conclusion from that test.  Let me explain :

1/ I got the nightly (21 march 15) build and installed it in the radio.
2/ I uploaded the previous version of my script (the one which didn't run anymore in 2.0.15) in my test model.
3/ Result : "syntax error".
4/ Then I tried the latest version of my script (the one which runs in 2.0.15).  It doesn't work either in the nightly build.  So I checked and saw that that build includes the heli support.

Someone reported to me once that the script didn't work on his T+ in 2.0.15.  After checking the noheli option, and reflashing, he got it working.

Did the script already break in 2.0.13 with the heli support ?  I have no idea :-(

Conclusion : I can't really compare if there's a difference between the official build and the nightly one, unless there's one with the noheli option.

mpaperno, 2015-03-28:
FWIW, I can confirm that setting NANO=NO gives more memory for Lua scripts on the hardware.  Or at least some scripts I have run OK, but run out of memory with NANO=YES.  I'm not sure how to quantify this further (check available memory) except perhaps systematically using more and more until it runs out.

OTOH I have found that pre-compiling script to bytecode can make a big difference.  The same scripts which run out of memory with NANO=YES, run fine on same firmware when pre-compiled.  I'm also showing a 25% reduction in memory reported after GC when running pre-comiled versions vs. plain-text.

I got the idea from this post: http://openrcforums.com/forum/viewtopic.php?f=45&t=6502  I've posted a couple messages to that thread, but since it's my first posting they're lost in moderation limbo. :(  My post includes a link to code for automatically generating compiled versions of scripts when the simulator runs them.

Anyway, just thought it might be relevant to this issue.  My scripts ran fine in the sim, but I ran out of memory pretty quickly on the hardware (T+) with 2.0.15.  Pre-compiling the scripts definitely saved the day. 

Cheers,
-Max

on4mj, 2015-03-28:
I guess YMMV.  When the problem arised, I tried to compile most of mine, keeping only the code which is meant to be user-customized and including the compiled one with a dofile().  Nothing changed.  But I didn't have the debug tool in companion at the time.

Is it possible to have a nightly build with noheli and nano=no so that I could do some tests (I don't have the toolchain to build the firmware and don't have time to gather all the pieces) ?  TIA.

bsongis, 2015-03-28:
Ok, this is what I expected. I will revert the default to NANO=NO

closed 2015-03-28 by bsongis

projectkk2glider, 2015-03-29:
This apparent lack of memory is caused by the behavior of *newlib-nano* allocator. The newlib-nano has very simple allocator that does not handle very well the (de)allocation pattern of Lua interpreter. I wrote more about this in https://github.com/opentx/opentx/issues/1643#issuecomment-54118746. 

To solve this problem and make more efficient use of available memory regardless of the allocator used, I wrote the *Bin Allocator* (#1804) which is only included in `next` branch unfortunately. But according to my tests the  *Bin Allocator* should really help with this issue.

Looks like I newer really explained how the  *Bin Allocator* allocator works:

###  How Bin Allocator system works

The basic idea for *Bin Allocator* is to reduce memory fragmentation by satisfying small memory allocations from preallocated fixed size memory chunks. The memory occupied by these chunks (bins) never gets fragmented.

*BinAllocator* class:
* Has an array of fixed size memory chunks. These are called *bins*.
* implements *malloc()* function that tries to satisfy it by returning one of the bins.
* keeps track of which bins are in use

*Bin Allocator* wrapper:
* implements standard allocator functions (malloc, free, realloc,...)
* creates two differently sized bin containers (these numbers are based on debugging and testing with real Lua scripts on radio):
  * 200 bins of 29 bytes 
  * 50 bins of 91 bytes
* For each malloc request it tries to satisfy it from one of the free bins. It chooses the bin based on requested memory size.
* If a malloc can't be satisfied from bins, it passes malloc request to system allocator

Integration in to existing code:
* Lua allocator is replaced by the  *Bin Allocator*

mpaperno, 2015-03-30:
I spent a bunch of hours with this...  First I ported the Bin Allocator changes to the master branch (https://github.com/mpaperno/opentx/commit/80cfa113e12bcf10680e9397c68dd92733792192).  Then I tested different size scripts against multiple builds: 

    A) nano=no  & bin_alloc=no
    B) nano=yes & bin_alloc=no
    C) nano=no  & bin_alloc=yes
    D) nano=yes & bin_alloc=yes

The different versions do allow slightly different sized scripts.  However the difference between each is minimal, a few lines of code either way.  I basically kept adding or removing lines until it ran out of memory on each build version, and kept copies of each script (before/after error).  Once I narrowed down these "trigger points," I repeated each test multiple times, reloading different firmware and script versions, and the results were consistent each time.

- Version A ran out of memory the quickest (allowed the smallest script).
- B came in second, allowing about 4 more long lines of text to be displayed on the LCD.
- C was the best, allowing 4 additional lines of long text, and about a dozen other short lines of code.
- Interestingly, version D ran out of memory at about the same script size as A.

So there seems to be some gain to be had with the Bin Allocator.  But the difference in allowed code size is almost academic (no offense! some improvement is better than none!).  By far the biggest gain comes from pre-compiling scripts... no memory issues in any of those firmware versions, with total running scripts almost twice as long as the ones tested above.

I also confirmed that comments make no difference, and that where you declare local variables doesn't matter (in function or outside of function).  Basically seems to boil down to the overall functional code size.  Examining the compiled bytecode in a hex editor can be quite revealing.  Also the output of `luac -l -p <script.lua>` .

Thanks to whoever approved my posts about script compiling over at http://openrcforums.com/forum/viewtopic.php?f=45&t=6502#p94915 

-Max

on4mj, 2015-03-30:
I might be wrong, but from what I understand, the "bin allocator" would make little difference in the size of code available itself (the parsing is probably done in an incremental way, so no fragmentation shouldn't occur at that point).

The gain would be more in run-time allocations (like filling/clearing tables or creating/destroying objects).

Anyway, your "loading compiler" might be a good idea.  That's how Python works, after all.

mpaperno, 2015-03-30:
Just to clarify, all the "out of memory" errors I have been running into so far are clearly during the script loading phase, not parsing or execution. The fw I'm running with the compiler also separates the loading error from the parsing error (https://github.com/mpaperno/opentx/commit/1ef83c69656a4fe3c43214d588fb5a4ab5700d94#diff-f49e438ac760aed50058e8f25e88649fR1690), so there's no confusion (needs a new status type so it can be properly reported to a user w/out debug info).

So it appears that the allocator is doing something more efficiently than "nano" even during loading.  Though you're probably right about the runtime gain when using it being more important, which I plan on testing next.

-Max

mpaperno, 2015-03-30:
OK, me again...  Tested running the same set of scripts with the different fw builds (same A/B/C/D as above).  Bear with me while I try to explain how I'm testing so you can decide how valid it is. 

What I have is one "main" script running as a MIXER, which parses out a bunch of custom telemetry fields ("abusing" things like RPM, T1/2, FUEL, etc) and makes the real values available to telemetry scripts as a "MAV object."  It also sets 3 outputs based on various statuses from the MAV.

Plain-text messages from the MAV are transmitted 2 chars at a time using the RPM field.  The mixer script parses these chars and stores them in a buffer until it sees an EOL.  Then it stores the full message in an array and clears the receive buffer.  The array has a finite length (15) so once it fills up the oldest message is replaced by the newest in a rotating fashion.  The messages are of varying lengths so the buffer and the array member sizes are constantly changing.  The array also holds a priority and timestamp for each message.

So then there are the actual telemetry screens which display some data and/or messages.  These are relatively simple.  They load the "MAV object" from the mixer script as a Lua table of variables and functions.

I load each firmware, then mark the GC reported memory use at startup.  Then I reset the flight controller, which sends a long stream of text to fill up the messages array.  I can watch the memory use go up. After the flood I make a note of the reported memory.  I then restart the FC a few more times, noting memory each time, and making sure there are no leaks.  I switch around between different screens as I do this also, but take readings always from the same custom telem screen.

What I found is very little difference between the firmwares.  In fact B, C, and D, reported almost the same numbers while running and the exact same number at startup.  The surprise "winner" of this contest, by a very slight margin, was A -- nano = yes, bin_alloc = no.

Here's my log... the numbers are bytes reported as "GC use" in debug console.  BTW I'm running debug version, which may or may not affect things, I don't know. The scripts are pre-compiled and are the exact same version in all tests.  One thing I did notice, as noted below, that the "nano=no" builds seems to run GC more often than the others.

	debug (nano)
	40695-startup
	41225
	41227
	41223

	binalloc-debug
	41001-startup
	41526
	41507
	41508
	41509

	nonano-debug   *gc happens more often
	41001-startup
	41529
	41555
	41548
	41533
	41529

	nonano-binalloc-debug     *gc happens more often
	41001-startup
	41534
	41551
	41509
	41555
	41461

	debug (nano) reloaded
	40695-startup
	41247
	41249
	41227
	41242
	41247
	41225

BTW, doing a similar exercise in the simulator shows a memory leak -- GC report just keep climbing overall.  Perhaps the very leak that #1804 fixed (haven't tried it yet).  

Well, don't know how much my finding are worth, but there they are.  Sorry for the length.

-Max

projectkk2glider, 2015-03-30:
@mpaperno very interesting stuff! I am a little disappointed at the *improvement* of Bin Allocator. I wished it would be more helpful. 

If you have time and will, you could play with the Bin Allocator's bins (their size and number of them). It is also helpful if you uncomment various TRACE macros that report allocation/dealllocation events. Then you can see the memory usage patterns of Lua and maybe came up with optimum solution.

>One thing I did notice, as noted below, that the "nano=no" builds seems to run GC more often than the others.

GC is done same number of times regardless of build options. But the memory usage after GC is reported only it it changed from previous value. So in that case used memory size changed more times, that is all.

>BTW, doing a similar exercise in the simulator shows a memory leak -- GC report just keep climbing overall 

It might be normal behavior since RAM is abundant on PC and GC is not so aggressive. Lua also triggers emergency GC if it runs out of memory (ie if it gets failed malloc() return). This happens a lot on radio, but almost never on PC.

mpaperno, 2015-03-31:
Yes, the traces reveal some interesting stuff.  It quickly became apparent why the runtime memory usage matched closely with the other versions -- the allocated bins fill up very quickly during initial script loading/parsing, and are then already full when the script itself is running.  Very few realloc calls actually hit the bins ("OUR realloc" in the traces) -- about a dozen out of hundreds which fall back to libc realloc.  Also, by default BA was only reserving about 10KB total in all the bins, and my scripts are using 40+... so clearly the bin allocation was just overwhelmed.

The other problem I found with the bins is the old "block size issue" still plaguing mass storage to this day.  The memory block sizes requested by Lua are all over the place... 20B to >500B.  Though many fall in the <100B range, maybe even <50B.   I tried playing with the size and number of bins, but with only 2 size options, there's just not enough variety to match the requests closely.  I thought of using more slots, but...

One could probably write a thesis on this subject.  :)  And I'm certainly no expert.  I guess at this point my question would be what are we trying to determine or optimize?  Intelligent memory allocation so that minimum RAM is used no matter what?  How much speed difference does aligning data chunks in RAM really make?  Is write speed important, or mostly read?  Are we OK with statically reserving RAM for Lua (like Bin Allocator does)?  if so, how much?  And again if so, does it really matter if data is "fragmented" within that reserved chunk of memory or not (requires testing to answer, I'm sure)?  (I'm not expecting answers necessarily, but showing where my mind goes with all this.)

The initial loading of a plain-text script clearly takes up "a lot" of memory up front, much of which is then freed up afterwards (apparently).  To me this seems to be the biggest stumbling block in OTx at the moment.  My previous findings with loading different sized scripts might simply be a result of total RAM actually available in the different fw versions, and nothing to do with the allocator used. 

I briefly checked (and might look some more) into how CHDK handles Lua scripts.  They seem to be just letting the default memory allocator (libc (re|m)alloc()) deal with it, but I've really only scratched the surface of that extensive code base.

Cheers,
-Max

kilrah, 2015-03-31:
> Plain-text messages from the MAV are transmitted 2 chars at a time using the RPM field. The mixer script parses these chars and stores them in a buffer until it sees an EOL. Then it stores the full [...]

Interesting. How much of a throughput do you get by doing things that way? How fast is the telemetry script at reading the buffer?

I'm currently looking at directly interfacing lua with both the smart port and the serial port, and that may help determining the required buffer sizes... We already have buffers for the serial port but will need to add 2 more for the smart port so lua access can be done separately from the main telemetry processing without disturbing it.

mpaperno, 2015-03-31:
> Interesting. How much of a throughput do you get by doing things that way? How fast is the telemetry script at reading the buffer?
>

I don't have any actual numbers for you... but that's a good question and I'll try to put some kind of chars/sec or other stats counter on both sides, out of curiosity.  The messages are buffered on the sender side as well, since they come in as full strings, not individual chars.  I can tell you that it works surprisingly well -- offhand I would say about at the speed of a 1200bps modem from the old days...  :)   In a long stream of messages (like during FC startup) I'll get a few garbled characters now and then, but mostly everything comes through clearly.  The message length is limited by the upstream MAVLink spec to 50 chars per msg, though most are much shorter.

> I'm currently looking at directly interfacing lua with both the smart port and the serial port, and that may help determining the required buffer sizes...
>

Well, my "buffer" is a simple script-local Lua var to which I append (..) the chars as they come in (using a table of printable characters indexed by ASCII code).  Once it sees a zero character, that's EOM and the "buffer" is cleared with an `= ""` .  The messages array (also script-local) member stores the whole message string as printable characters, until a new one comes along to replace it.  Changing the size of the storage array does affect memory usage of course, but not drastically (I could get you numbers if that would help).  I haven't tried pushing it past 15 stored messages.

I'll be posting the code somewhere, at some point, but I'm happy to send it to you if the relevant bits you want.  It would be fairly simple to replicate/integrate the "message sender" routine for testing into some other program that already sends SPort packets.

> Well, one thing is that a digital cam has a good 2-3 orders of magnitude more RAM available so I doubt they need to worry much about it...

Yea that would help matters.  :)

-Max

lshems, 2015-06-12:
I have an other problem which perhaps ius related and might suggest there are other issues.
The script from on4mj actually loads well now the first moments, but crashes when loading new code in the same local variable several times. Thgevscipt was designed to load the code for different tasks one at a time.

Even when assigning nil to the used vqariuable, and doing a GC, the script crashes for no apparent reason.

The script files can be downloaded here: http://www.rcgroups.com/forums/showthread.php?t=2298914&page=11

Guido

commit 21807325 (2015-03-28): Fixes #2050
commit 3c7d7032 (2015-03-28): Fixes #2050