opentx/opentx #2054: Ability to Adjust 'Cells" voltage value per model

dave-pitman, 2015-02-03:
*Sorry if this is addressed somewhere else.*

Using a Taranis+ with a Naze FC and a D4R-II rx, the pack voltage is reported to Taranis as "cells".
This voltage is often off ~ .2v which is significant.
There needs to be available a way to calibrate this value in a similar manner as the Tx's own battery value can be adjusted.
It would probably be on a per-model basis.

Thanks Guys!

bsongis, 2015-02-03:
I don't get why Naze sends this value as "cells" while there is only one sensor. It should be sent as "vfas".

closed 2015-02-03 by bsongis

dave-pitman, 2015-02-03:
It was suggested that I change my voltage input from 'cells' to 'fvas', which I did.  But, the same issue there.  There is no way to adjust or scale the value to correct it.  'cells' or 'vfas'  if it is not accurate in Taranis (compared to mulitmeter, not fc board) then we need a way to calibrate, yes?

bsongis, 2015-02-03:
Perhaps it could be calibrated on the Naze! By the way there is an offset in Taranis, so it's possible to use it.

dave-pitman, 2015-02-03:
The voltage displayed in Naze (using Cleanflight fw) is correct. (It matches the multimeter).  It you adjust in Cleanflight until it is more accurate in Taranis, then it is incorrect for anything else reading the Naze, like OSD for example.

It needs to be adjustable in Taranis.  I tried changing the fas offset value, and for me it did not change anything.  For example, I put in a value of 10.0 and nothing changed on the vfas screen.
Am I doing something incorrectly?

dave-pitman, 2015-02-03:
Kilrah is saying that the 'fas offset' does not apply to voltage, only current.

bsongis, 2015-02-03:
If the value is correct on Naze, why would it be different on Taranis? There is no conversion here in OpenTX!

dave-pitman, 2015-02-03:
I do not know.  Why does A1 need adjusted to be accurate in Taranis /
OpenTx ?

Dave

dl@pitman.us  |  360.460.4200

...sent from some device or another...;-)

bsongis, 2015-02-03:
A1 depends on resistors, it's a 256 levels value, the unit coming from the sensor is not Volts, while VFas is.

projectkk2glider, 2015-02-04:
VFAS problem is explained in #1986

dave-pitman, 2015-02-04:
I understand from this and #1986 that there is no interest in providing adjustment to the incoming voltage signal within OpenTx, correct?

You say that the signal should contain the correct voltage, and that adjustment needs to be made on the incoming (cleanflight) side.  But I ask you to explain why the displayed voltage is correct in the Cleanflight GUI AND on an OSD device reading the same stream as the OpenTX VFAS.

Do you think that the data stream should be adjusted to be incorrect to all other devices so that OpenTx reports the correct value?  That just doesn't seem reasonable or desirable.

Apparently Mike Blanford has written some scaling adjustments for ersky9x that they are going to see if they can be used to adjust the voltage value on the radio.

I appreciate the work you guys do, that I am not capable of, Thanks!

@projectkk2glider  Thanks for the link to the other ticket. I searched for similar tickets, but admit I didn't search the closed pile.

bsongis, 2015-02-04:
Yes this calibration will be possible, but in OpenTX 2.1, not in 2.0 (it's already implemented, you can try the nightly builds)

projectkk2glider, 2015-02-04:
I have now looked at the Cleanflight source code and found out:
* the VFAS sending was removed in and then added  back in 5b51780780124efc78296cb9112202e76415827a

projectkk2glider, 2015-02-04:
I have now looked at the Cleanflight source code and found out:
* the VFAS sending was removed in and then added back in commit https://github.com/cleanflight/cleanflight/commit/5b51780780124efc78296cb9112202e76415827a (well before version 1.0)
* it uses the standard and imprecise way of sending VFAS, (ap and bp values).

There were no relevant changes to frsky telemetry code since then. 

Cleanflight could change VFAS sending to VFAS_ID  (0x39) which will then give true 0.1V resolution. I am still looking if there is any other reason for voltage discrepancy. I will report this to Cleanflight project and maybe create my first contribution over there.

projectkk2glider, 2015-02-04:
The voltage sending via CELL values is also far from optimal in Cleanflight. I will also report that.

kilrah, 2015-02-05:
Imo the cell thing should be removed altogether. I even think it was only added as protocol test more than anything else.