opentx/opentx #2056: [enhancement] Set RTC when receiving GPS timestamp

hsteinhaus, 2015-02-04:
The Taranis RTC has the habit to drift badly (at least mine). Wouldn't it be possible to correct it when receiving GPS timestamps via telemetry?

LapinFou, 2015-02-05:
Good idea ! :+1: 
It would also be nice to update the RTC, when some models are uploaded in the radio from a PC/Mac.

bsongis, 2015-02-05:
RTC from GPS is nice with me. But from the USB connection we have to change the protocol, massstorage is not good for that (it's not stored in EEPROM).

I would need some inputs about which USB protocol would be better to our needs. Massstorage showed us some limits for EEPROM and FIRMWARE transfers ...

LapinFou, 2015-02-05:
OK. It didn't understand than firmware/model update where done through a mass-storage protocol.
Updating RTC from GPS should take care of timezone. I guess this easiest way will be to only update minute/second (with special caution if the minute are closed to a plain hour) .

bsongis, 2015-02-05:
No, timezone is configured in General Settings.

LapinFou, 2015-02-05:
Sorry, I did not mean timezone (bad English), I mean Summer TIme/DST (Daylight Saving Time). I'm not at home, but I don't recall such setting in OpenTx.

kilrah, 2015-02-06:
Just add / remove one to the timezone adjustment...

LapinFou, 2015-02-08:
Thanks for the tips. Maybe it could be automated?

kilrah, 2015-02-09:
Nope, no way of knowing for sure when it happens. Devices that do use network access.

hsteinhaus, 2015-02-09:
I think the best solution would be to run the RTC in UTC (or more precisely GPS time) and a simply adding an adjustable offset before displaying the time somewhere in the menu. 

BTW, telemetry logs should  preferably contain the GPS time (not local time) for easier matching with onboard vehicle logs, e.g. provided by a multicopter/fixed wing autopilot/stabilizer.

JoeB99, 2015-02-15:
I think the RTC and all logs should be in UTC, then apply the user-specific offset (as set in general settings) to the displayed time only. 
Automatic DST change to the time can be done on the radio. There is a proven formula to calculate when DST occurs. 
The user could set their DST amount (in 15 min increments) in a general settings menu. For most, this would be set to 1hr.

jtaylor2, 2015-02-15:
>Automatic DST change to the time can be done on the radio. There is a proven formula to >calculate >when DST occurs.
  
OpenTX is used all over the world.  Daylight savings time is used (or not used) differently in different countries and can change at the whim of an administration.  http://en.wikipedia.org/wiki/Daylight_saving_time_by_country 

I think you underestimate the complexity of automatic DST change.  I would be interested in a pointer to "a proven formula to calculate when  DST occurs" because the only way I know how to do it is with the zoneinfo database ( http://en.wikipedia.org/wiki/Tz_database ) which seems like it would be too involved for an embedded system without OS TZ support.

Is it really that much trouble to change the time on the transmitter twice a year?

kilrah, 2015-02-15:
Even my fancy radio-controlled watch that supports displaying 3 timezones at the same time requires you to set DST manually for each zone because it's impossible to know for sure when it's active. The "rules", when some exist, are regularly changed.

JoeB99, 2015-02-16:
Yes, you need to know the day and the offset. Companion connects to the internet to get the latest firmware version. It could also get the rule for the users timezone at the same time, and then send the rule to the radio.... Or just let the user change the time 2x a year :)

kh4, 2015-02-16:
Maybe we can just make GPS correct time for +/- 5 minutes or so? This way it could just guess the timezone?

JoeB99, 2015-02-16:
That won't work for me, my TX didn't come with a GPS receiver :-P

kh4, 2015-02-17:
@JoeB99 none of the TX:s have GPS, this feature will update TX time based on GPS data received via telemetry from the aircraft/rover.

JoeB99, 2015-02-17:
It was a joke, hence this-->      :-P

commit 00c4a527 (2015-02-17): Fixes #2056 - Needs tests!
commit b40a06ef (2015-02-20): Fixes #2056 - Companion part
commit 2a3c5958 (2015-03-22): Fixes #2056 - Needs tests!
commit c44a3555 (2015-03-22): Fixes #2056 - Companion part
closed 2015-03-22 by bsongis