opentx/opentx #2058: an idea for calibrating altimeter with every launch

nigelsheffield, 2015-02-06:
http://www.rcgroups.com/forums/showthread.php?t=1866206&page=2673

Would it be possible to be able to read presure  and set the reference presure for zero altitude in a lua script?
This would make it possible to detect a launch and then go back to lowest presure reading prior to launch and set that as the zero reference.
Thus keeping the altimeter from drifting.

kilrah, 2015-02-06:
All the sensors that are commonly used along with the Taranis do not transmit a pressure, but the "ready" altitude they calculate themselves. There isn't even a "pressure" item in the FrSky telemetry protocol.

And by the way, what you describe is no better than doing the same with the altitude reading, both will drift the same.

closed 2015-02-06 by kilrah

nigelsheffield, 2015-02-06:
So by detecting a launch and looking back at the last readings to find the lowest point and using  that offset  against a second altitude in the script would be possible anyway.
Thanks.