opentx/opentx #2062: Mix name is cropped

JoeB99, 2015-02-10:
Comparision 2.0.15 - Win7, x64

Using the wizard, create an airplane, and add flaps on 2 servos.

Go to mixes. The Names are (Flaps Up) and (Flaps Down).

Edit the mix. 

Only 8 chars are allowed for the mix name, so "Flaps Do" is displayed.

'Inputs' names are limited to 4 chars.
'Servos' names are limited to 6 chars.
'Curves' names are limited to 6 chars.
'Mixes' names are limited to 8 chars,
'Special Functions' don't allow names at all(?)

It would be nice to see consistency here, e.g. everything limited to 8 char...

tomkfly, 2015-02-10:
There isn't enough memory, so something else would have to be limmited. The names were shortened to allow 32 logic switches instead of the 16 we had previously. I know which I prefer. You just need to learn how to abreviated and don't wast a character by having a space.

Tom

JoeB99, 2015-02-10:
Ahh i see. So is the SD Card not available to store settings?

kilrah, 2015-02-10:
No, SD card is only for non-essential things. It must be expected it will not be available in case of unexpected reboot.

But it sounds like default names in the wizard script might need adapting to the new length.

Actually, did you re-download the current script? Could have been updated since you loaded it.

tomkfly, 2015-02-10:
The SD card is not able to run as part of the process. The EEPROM is loaded into the onboard system/ processor memory. The SD card can store copies of EEPROMS, but they have to be written into the processor memory to be used.

Tom

JoeB99, 2015-02-10:
Sorry guys, im new to this. I downloaded the latest comparison v2.0.15 from the website yesterday and used that to create a new model.
Is it not possible for only the required part of the "eeprom" from the sdcard to be read at a time? e.g. only when displayed to the user?

Stuff like names etc that are only for usability display to the user could be stored in an 'extended' struct, and only read on demand, rather than loading the entire file to ram.. maybe?

kilrah, 2015-02-10:
Too complicated to maintain, problems with needing to upgrade the SD card files along with the firmware on subsequent updates, unability to read both together from companion...

kilrah, 2015-02-10:
And the wizard is separate from both the firmware and companion, needs to be downloaded and loaded onto the SD card separately.

commit d222e0c8 (2015-04-01): Fix #2062
commit 96fa3939 (2015-04-01): Fix #2062
closed 2015-04-01 by kilrah