opentx/opentx #2063: Telemetry issue with A1

Edgs, 2015-02-10:
There appears to be an issue with A1 telemetry values (and possibly others - I didn't test any further)
Summary:
Occasionally, after the receiver is powered off & on, causing the transmitter to lose then recover telemetry, a persistent offset is introduced to the read A1 voltage, which never recovers until the transmitter is power cycled.

Transmitter:
Taranis B (2.0.15) with internal module (XJT_141016.frk)
Receivers:
X4R (X4R_150116.frk)
X8R (X8R_150115.frk)

All firmwares were updated from previous versions (including the Taranis from pre 2.0.x) and the problem has always existed on these units.  I thought that the receiver firmware updates might solve the problem as I have seen mention of voltage measurement issues.

Details:
When first powering the receiver, the returned A1 value seems to always be correct, for example 4.97V.
When powering off the receiver, *if the reported A1 voltage drops at the same time as the telemetry is lost* then when repowering the receiver, the displayed A1 voltage will be half way between the dropped voltage, and the correct expected voltage.  For example, with the X4R, the voltage seems to nearly always drop to 4.60V when powering off (and stays at that voltage in the log while telemetry is lost) and once recovered, only returns to 4.81V (mean of 4.60 & 4.97?).  It will then stay at 4.81V permanently, and never 'recover' to the expected 4.97V.
It does appear to still be being updated, as powering off the receiver does still occasionally cause the voltage drop, with the accompanied ramp-down of the displayed voltage due to the averaging.
It displays this behavior very often, maybe 50% of the time, and it is very consistent - in the above example, the voltage will *always* end up at 4.81V.

Is it possible that the 'glitch filter' is stopping the voltage from recovering to it's full value after receiving an erroneous low voltage, and a period of lost telemetry?

projectkk2glider, 2015-02-11:
The *glitch filter* is a simple averaging of the last N values, see https://github.com/opentx/opentx/blob/master/radio/src/telemetry/frsky.cpp#L79

My bet is that receiver is sending wrong values.

Mow06, 2015-02-11:
Sounds to me like someone is looking at A1-   not A1

Edgs, 2015-02-11:
What is A1- ?  The value I am referring to is the internal supply voltage of the RX, which is reported in the Taranis telemetry as A1.

It's possible it is the RX at fault, but once the 'bad' value exists, powering off & on the RX never recovers it, whereas powering off & on the TX always recovers it, which points to it being something in the TX.  Also, the first time the RX powers up, it's always correct.  As far as the RX is concerned, every power on should look the same.  It's the TX which is staying on as the fault appears.

I'd be happy to test further, and can grab some datalogs if required.  I'll also have a quick look at the code.  A fresh pair of eyes etc.

tomkfly, 2015-02-11:
A1- would be the lowest recorded value of A1

Edgs, 2015-02-11:
I made a short 1Hz datalog to demonstrate the problem.  I commented the significant points.
See RSSI for the connection state, and A1 with the problem.  Cell volts included for comparison.
Looking at this it seems to have received a the last A1 packet with a low voltage in, just before the RSSI alarm.

```cpp
Time 	RSSI 	A1		A2	Cell volts
21:17:01.78	0	0		0	0
21:17:02.74	0	0		0	0
21:17:03.75	0	0		0	0
21:17:04.76	0	0		0	0
21:17:05.78	0	0		0	0
21:17:06.78	0	0		0	0
21:17:07.79	0	0		0	0
21:17:08.80	84	4.96	0	0		// First connect here
21:17:09.81	81	4.96	0	11.3	// all good
21:17:10.82	81	4.96	0	11.3
21:17:11.84	83	4.96	0	11.3
21:17:12.85	86	4.96	0	11.3
21:17:13.87	82	4.96	0	11.3
21:17:14.88	83	4.96	0	11.3
21:17:15.88	82	4.96	0	11.3
21:17:16.89	83	4.96	0	11.3
21:17:17.90	81	4.96	0	11.3
21:17:18.91	83	4.96	0	11.3
21:17:19.91	86	4.96	0	11.3
21:17:20.92	86	4.96	0	11.3
21:17:21.93	0	4.96	0	11.3	// disconnected
21:17:22.94	0	4.96	0	11.3
21:17:23.96	0	4.96	0	11.3
21:17:24.97	0	4.96	0	11.3
21:17:25.98	0	4.96	0	11.3
21:17:27.00	0	4.96	0	11.3
21:17:28.01	0	4.96	0	11.3
21:17:29.03	86	4.96	0	11.3	// reconnected
21:17:30.03	86	4.96	0	11.3	// still good
21:17:31.04	85	4.96	0	11.3
21:17:32.05	81	4.96	0	11.3
21:17:33.06	79	4.96	0	11.3
21:17:34.06	81	4.96	0	11.3
21:17:35.07	85	4.96	0	11.3
21:17:36.08	84	4.96	0	11.3
21:17:37.09	86	4.60	0	11.3	// ! receives low A1 as disconnecting!
21:17:38.09	0	4.60	0	11.3	// disconnected
21:17:39.10	0	4.60	0	11.3
21:17:40.10	0	4.60	0	11.3
21:17:41.10	0	4.60	0	11.3
21:17:42.11	0	4.60	0	11.3
21:17:43.12	0	4.60	0	11.3
21:17:44.12	82	4.81	0	11.3	// reconnected
21:17:45.14	81	4.81	0	11.3	// gone bad
21:17:46.14	83	4.81	0	11.3
21:17:47.15	83	4.81	0	11.3
21:17:48.16	82	4.81	0	11.3
21:17:49.17	83	4.81	0	11.3
21:17:50.18	85	4.81	0	11.3
21:17:51.19	82	4.81	0	11.3
21:17:52.20	81	4.81	0	11.3
21:17:53.20	82	4.81	0	11.3
21:17:54.21	82	4.81	0	11.3	// disconnected
21:17:55.22	0	4.81	0	11.3
21:17:56.23	0	4.81	0	11.3
21:17:57.24	0	4.81	0	11.3
21:17:58.25	0	4.81	0	11.3
21:17:59.27	0	4.81	0	11.3
21:18:00.27	59	4.81	0	11.3	// reconnected
21:18:01.29	84	4.81	0	11.3	// still bad
21:18:02.30	82	4.81	0	11.3
21:18:03.30	83	4.81	0	11.3
21:18:04.31	87	4.81	0	11.3
21:18:05.32	87	4.81	0	11.3
21:18:06.33	87	4.81	0	11.3
21:18:07.34	85	4.81	0	11.3	// disconnected
21:18:08.35	0	4.81	0	11.3
21:18:09.36	0	4.81	0	11.3
21:18:10.37	0	4.81	0	11.3
21:18:11.38	0	4.81	0	11.3
21:18:12.38	0	4.81	0	11.3
21:18:13.39	0	4.81	0	11.3
21:18:14.40	0	4.81	0	11.3
21:18:15.40	81	4.81	0	11.3	// reconnected
21:18:16.41	80	4.81	0	11.3	// still bad
21:18:17.42	78	4.81	0	11.3	// never recovers
21:18:18.43	82	4.81	0	11.3
21:18:19.43	84	4.81	0	11.3
21:18:20.44	89	4.81	0	11.3
21:18:21.45	91	4.81	0	11.3
21:18:22.46	85	4.81	0	11.3
21:18:23.47	79	4.81	0	11.3
21:18:24.48	82	4.81	0	11.3
21:18:25.49	83	4.81	0	11.3
```

Mow06, 2015-02-12:
I can confirm this. Just tried it on an X8R. Took about thirty cycles to see it.
A1 voltage of 6.52v 
6.47 on x8r shutdown.
6.41 when x8r repowered .
Power cycled Taranis.- Still 6.41 on reconnection.
Power cycled x8r - still 6.41.

Repower x8r with 5v supply, (Taranis untouched.) A1 shows 5.2V OK.
Repower x8r with original 6.52v (Taranis untouched) A1 shows 6.52v as original. 

Seems to indicate a RX related problem. Don't understand how it survives a power cyle.

Mow06, 2015-02-12:
Tried to analyse S-Port data, but haven't been able to create condition again.

mpjf01, 2015-02-12:
I tried to do this test and can't see any change in A1. However it may be
relevant that if A1 is set to display as "BARS" the value does not reset to
zero when the Rx is powered off yet the "NUMS" screen A1 value reverts to
zero when the Rx is powered off.

Edgs, 2015-02-12:
I'm not at all familiar with the code, but one possible thing I noticed is that s.port messages ``ADC1_ID 0xf102`` and ``BATT_ID 0xf104`` both appear to be put into ``frskyData.analog[0]``.  If that is the case, and both of these are being received alternately, and with different values, you would end up with a value half way between the two.
Unfortunately I don't have a build environment set up to debug.
Incidentally, resetting telemetry doesn't clear the condition, but sets the min & max to the persisting bad value.
Mow06 - I have never seen this survive a TX power cycle myself, but then again I seem to get it on 30-50% of RX power cycles.  Maybe  I'm just unlucky!

kilrah, 2015-02-15:
> I'm not at all familiar with the code, but one possible thing I noticed is that s.port messages ADC1_ID 0xf102 and BATT_ID 0xf104 both appear to be put into frskyData.analog[0]. If that is the case, [...]

These shouldn't be able to come at the same time. ADC1_ID would be fed only by D series receivers, and BATT_ID only by X series receivers (each not supplying the other obviously).

According to @Mow06 's testing it seems not to be an OpenTX issue. Might be worth sending your data to FrSky though, maybe they can fix something.

closed 2015-02-15 by kilrah

Edgs, 2015-02-15:
Just to confirm, this is not an OpenTX issue.
I managed to log some s.port messages during the event, and they do reflect what is seen on the TX display, so the data is already bad when OpenTX gets it.  It is a problem with the BATT_ID value.
Due to the circumstances of the issue, it seems most likely to be an XJT problem, rather than a receiver problem.  I will try to raise this with FrSky.
Thanks for looking a it.

kilrah, 2015-02-16:
Cool, thanks for following up!