opentx/opentx #2064: LUA direct access to smartport data

janekx, 2015-02-11:
I am looking for solution how to directly read and send data from and to smartport by Taranis. 
In smartport is some reserved space and I would like to use it to control and check status of the device connected to RX smartport by LUA script.
Now LUA reads only parsed datas from telemetry and it is useless for my project. Is possilbe to add LUA to smartpot bridge to send and recieve a few bytes?

kilrah, 2015-02-11:
There's already an open issue for that. I'm on mobile so I don't have the number at hand.

janekx, 2015-02-11:
Please can you send me that issue when you will be online again? Cant find it. Thanks

kilrah, 2015-03-11:
Can't find either, maybe it was never actually opened as an issue but it is on our list.

projectkk2glider, 2017-01-14:
I am pretty sure this can be closed, done in #3326 and 862fa5525ebf14a84a33c090ac89ca8cb77aafba and #4205

closed 2017-01-14 by projectkk2glider