opentx/opentx #2065: DSM2 subtype setting (DSM2 vs. DSMX) has no effect

zieren, 2015-02-11:
I'm using a DSMX module reportedly taken from a DX5e (bought it on ebay). The DSM2/DSMX setting seems to have no effect - I believe the module always uses DSMX (see http://www.rcgroups.com/forums/showpost.php?p=30756168&postcount=16135 for how I tried to tell DSM2 from DSMX w/o a spectrum analyzer).

Looking at dsm2_arm.cpp I see that the DSMX_BIT is being used. It's the default block of a switch though - maybe the switch value isn't set?

In pulses_avr.cpp DSMX_BIT is also defined, but never used. I don't understand what it does, I'm just saying :-)

bsongis, 2015-02-11:
You even don't tell us which radio you have!

zieren, 2015-02-11:
Sorry, I only mentioned that in the linked post. Taranis plus w/ 2.0.13. I re-checked just now by power-cycling the RX (AR6210) and checking whether its LED blinks (DSM2) or goes back to solid (DSMX), as was suggested on RCG. This now indicates it's using DSM2. The module is configured as DSMX in the Taranis.

That changes things a bit: Either the Taranis doesn't set the DSMX_BIT, or the module doesn't support DSMX. The latter is definitely possible. I didn't desolder it myself, just bought it on ebay. The seller told me it's from a DX5e, and it does match all the pics on the web I found, but maybe it's just very old or something.

What about DSMX_BIT not being used in pulses_avr.cpp? Sorry about the high level nit - I'm not too familiar with embedded systems so it might be perfectly common.

-Jörg

zieren, 2015-02-12:
The seller assured me that the module is from a DSMX capable DX4e or DX5e. When I have some time I'll put it into an original DX4e and see if that makes it do DSMX, in which case the bug would be in OpenTX.

Since I need to unsolder etc. for that, an easier way would be if someone with the same type of module can confirm that DSMX works for them. I'll try to find someone and get them to do a test.

Alexander-C, 2015-02-12:
I have a DX5e module in mine. With the transmitter set to DSMX I bound a HKR615x receiver. I followed your test of unplugging power from the receiver and reconnecting, the light on the receiver was solid.

I then set the transmitter to DSM2 and rebound the same receiver and did the same test as above and the LED on the receiver blinked...Hope this helps.

kilrah, 2015-02-12:
> I was under the impression that the module detected that the receiver was DSM2 or DSMX during the bind process and used whatever protocol was available.

Correct. You cannot force DSMX, you can only "allow" the module to use it if it can negotiate it with the receiver.

zieren, 2015-02-12:
Alexander-C, thanks a lot for trying this. What radio and software version did you use?

Right, the RX and TX negotiate DSMX vs. DSM2. But in dsm2_arm.cpp:162 OpenTX sets a flag if and only if DSMX mode is selected. I assume this flag causes the module to enable DSMX, and when it's not set the module will only use DSM2. This option is available in Spektrum radios, so it seems plausible that the module offers this.

I will build another module with a guaranteed DSMX capable donor (though it might be a few days before I find the time), and will also ask more people to try it. Though the above result does seem to indicate that the problem is with my module, not with OpenTX.

I might also try to retrofit the module in question in a DX4e and see if that changes anything.

Alexander-C, 2015-02-12:
I have an 'A' version Taranis transmitter running OpenTX Ver. 2.0.15

zieren, 2015-02-14:
I flashed from 2.0.13 to 2.0.15 and it still doesn't work for me. I'll try the module in a DX4e next.

zieren, 2015-02-14:
The problem was with my module. I've replaced the module with one I removed myself from a DX4e and now DSMX works. Then I put the problematic module into the DX4e that used to do DSMX before and now it only does DSM2.

Also, setting the subtype to DSM2 on the Taranis does have the desired effect, i.e DSMX and DSM2 work. So I'm closing this, sorry for the churn.

closed 2015-02-14 by zieren