opentx/opentx #2066: Taranis and Phoenix flight sim. v4 updated to v5-Phoenix will not allow aeroplane  sims.

Trianti, 2015-02-13:
Phoenix will only see Taranis as being in helicopter mode,and am unable to alter, or reset Tx.Can anyone help please.

zieren, 2015-02-14:
What is "helicopter mode"? In Phoenix, can you try going to "System" -> "Your Transmitter" and see if the bars are moving when you move the sticks? If the bars are moving then it's a configuration problem with Phoenix. I use Phoenix myself with the Taranis but I don't understand what exactly your issue is.

kilrah, 2015-02-15:
Not an OpenTX issue in any case.

closed 2015-02-15 by kilrah

Trianti, 2015-02-17:
It is

ckleanth, 2015-02-17:
Trianti with all due respect no its not an opentx firmware issue. this is purely something wrong on your end. if there are channels moving within phoenix then your setup is working (hence no hardware issue) and this is purely a configuration problem within Phoenix.

zieren, 2015-02-17:
FTR, OpenTX 2.0.15 on a Taranis+ works fine with Phoenix on my machine. Mail me at <expires-03-2015@zieren.de>, I'd be happy to help if I can.