opentx/opentx #2071: [Wish][Companion]LUA wizard running within Companion

Sacre100, 2015-02-15:
LUA wizard run within the Tx or the simulator but not within Companion.

I think that having similar LUA wizard with graphical user interface would be great.

kilrah, 2015-02-15:
Companion has its own, more complete model wizard...

closed 2015-02-15 by kilrah

Sacre100, 2015-02-15:
Well, disappointing answer.

Companion wizard are limited and not expandable, For example, on airplane, limited to four mobile surface on wing, no snapflap, no dual rate, no exponential, only one servo for elevator, etc...

Like for telemetry, model or function script, adding LUA wizard within companion will allow creative end user to express themselves and it would benefit to OpenTx.

kilrah, 2015-02-15:
The companion wizard is frozen within a given version of companion, but apart from that is just as evolutive. 
If we supported Lua scripts in companion someone would have to write a new wizard for it, what's the point when he can already expand the existing wizard instead? Just a different language.

Sacre100, 2015-02-15:
Why unique capability like F3K training, F5J training or F3F practice were not developped with C/C++ for OpenTx before the arrival of LUA ?
Why wizard for multi-rotor is available on the radio but not in Companion ?
Why Smooth Ramping Tilt for camera exist, which is also to my knowledge, a unique feature not available on any other transmitter ?

Answer are obvious.

kilrah, 2015-02-15:
Because it's all done by volunteers, and they will do what they're interested in.

Anybody's welcome to provide code for other types of models, again language doesn't matter so they can do it in the existing canvas.

kilrah, 2015-02-15:
Because it's all done by volunteers, and they will do what they're interested in when they have time. Someone started doing the companion wizard, and stopped before it was finished. Just like someone else later started working on the lua wizard, which is incidentally not "finished" either because they got sidetracked too.

Anybody's welcome to provide code to make them better or for other types of models, again language doesn't matter so they can do it in the existing canvas(es).

And we actually don't have smooth ramping for cameras, it just happens to be possible with the standard OpenTX features.

Sacre100, 2015-02-15:
OK, understood, however, smooth ramping for cameras already exist since October last year, a quick google search and you will find it immediatly.

kilrah, 2015-02-16:
Well... Cool, I didn't know anything about it. So "we don't have smooth ramping" as the developers never made something specifically for it, someone made used the tools we offered them (lua) to make it.