opentx/opentx #2076: Turning variometer's sound signaling on and off

robert48t, 2015-02-15:
Welcome,
I'd like to know if you plan to add the ability to turn on and off the variometer's sound signaling when the model is not flying. This problem mostly affects the pilots of RC Gliders.

kilrah, 2015-02-15:
Define "not flying". 
Then from that definition (for example altitude >5m) create a logical switch that will turn the vario on only when that condition is true.

robert48t, 2015-02-15:
Vario informs about the vertical velocity of the model, so you can tell if it's going up or down. But when the model is on the ground, and it's not moving (it's in static position), or it's flying from point A to point B with a constant altitude (so called "insensibility zone") there's no need for the vario to give the sound signals. Only going up or down should be signalised with intermittent sound signal, with the frequency proportional to the vertical velocity.

Insensibility zone:
[image]
When the RC Glider is flying from point A to B (distance doesn't matter) with a constant altitude, it's in the "insensibility zone" and the vario shouldn't give any sound signals.

It would be fantastic if you could set "quiet zone" - the min and max altitude where the vario stays quiet, and gives no sound signals.
[image]
I marked with red color two altitudes (-0.3m and 0.3m, example configuration). The vario should stay quiet when the model's altitude is between those two. 

Hope now you understand :)

jtaylor2, 2015-02-15:
What you are asking for is the way it used to be.  It was changed to the way it is now at the request of some users.  See forum http://openrcforums.com/forum/viewtopic.php?f=86&t=5081&p=90360&hilit=vario+tones#p90360 .  As Kilrah pointed out, the old way of having a dead band can be created by using two logical switches and special functions to turn on vario only when wanted::

L1  a>x   vspd     0.3 m/s  and (a switch if wanted)
L2 a<x    vspd   -0.4 m/s   and (a switch if wanted)

SF1   L1   vario
SF2   L2   vario

or only one if the values needed are the same

L1 |a|>x   vspd   0.3 m/s
SF1   L1  vario

Jim

bsongis, 2015-03-18:
I reopen this one as it's something that will be configurable in 2.1. A lot of users are asking for it.

closed 2020-07-08 by stale[bot]