opentx/opentx #2078: Lua script values not available.

Datospel, 2015-02-15:
Some calculated values like dist, alt+, alt- etc are unavailable in Lua scripts. This causes some limitations when trying to develop some new telemetry scripts.

Is there a chance to open these values to Lua getValue function call?

Thank you

kilrah, 2015-02-16:
> Some 

Which ones exactly?
AFAIK individual cells are the only thing that is not availalble.

Datospel, 2015-02-21:
Dist, for example. It is shown when chosen to view on telemetry screen but in Lua if using getValue("distance") it returns 0.

projectkk2glider, 2015-02-21:
Well, according to http://downloads-20.open-tx.org/firmware/lua_fields.txt the **distance** should work, unless there is a bug.

Datospel, 2015-02-21:
When I choose it to see on one of 3 in-built telemetry screens - it is shown. 

When I am trying to get it inside Lua with getValue  - it's not. But I've only tried it with the name "distance". Didn't try to use 211 instead. 

Seems to be a bug. Other values are working fine. And no logging of Dist in log file.

projectkk2glider, 2015-02-21:
When I run this script in simulator:
```lua
local function run(event)
  local info = getFieldInfo('distance')
  print("id: "..info.id.." name: "..info.name.." desc: "..info.desc)
  local dist = getValue('distance')
  print("distance(by name) = "..dist)
  dist = getValue(info.id)
  print("distance(by id) = "..dist)

  print("pilot position: "..getValue('pilot-latitude')..", "..getValue('pilot-longitude'))
  print("model position: "..getValue('latitude')..", "..getValue('longitude'))
  return 1
end

return {run=run}
```

I get:
```
id: 212 name: distance desc: GPS distance [meters]
distance(by name) = 99
distance(by id) = 99
pilot position: 44.029516666667, 10.114786666667
model position: 44.029091666667, 10.115888333333
Script finished with status 1

```

It seem to work and distance is also correct according to http://boulter.com/gps/distance/?from=44.029516666667%2C+10.114786666667&to=44.029091666667%2C+10.115888333333&units=k

Datospel, 2015-02-21:
strange. thank you. I'll try it again tomorrow.

projectkk2glider, 2015-02-28:
LUA individual cell voltage is already requested #2051, other are non-issue. Reopen if needed.

closed 2015-02-28 by projectkk2glider