opentx/opentx #2079: in the Servos Screen can we have Trims => PPM Centre?

ckleanth, 2015-02-15:
as per title can we please have in the Servos Screen an additional option to transfer Trims value into PPM Centre shift? The reason is that if you use offset you end up with unequal servo travel and if you use symmetry then the endpoints is clipped.

if this is implemented it is suggested the function to:
- to zero any offset value and transferring that as well in the PPM centre shift.
-  the min and maximum servo endpoints to be moved according to the new PPM centre shift (to avoid binding)
Thanks

kilrah, 2015-02-16:
Mmh no. Neither limits nor PPM center should be moved automatically. Both are set according to the model's mechanics at the beginning of your setup and shouldn't be touched anymore.

Subtrim with symmetry is the solution. If you've followed the setup guidelines you have enough margin between desired throw and limits before it starts clipping.

closed 2015-02-16 by kilrah

projectkk2glider, 2015-02-16:
Also duplicate of #1952

ckleanth, 2015-02-16:
The issue is for 3d planes i use relative long throws. the ppm centre is used to ensure the servo arm is perpendicular to the control surface its connected to and the end points to ensure the control surface will not cause servo binding, Obviously one needs to choose a servo arm long enough to you try utilise all servo range available within the throws you need for the plane.
If you then trim the plane you have a small offset from the ppm centre but then the throws are asymmetrical by default. If you then you select symmetric throws then unavoidably the throws on the side you have the offset is clipped. With large control surfaces and expo then the end point clipping can be rather large.
My solution is the same as with #1952, I look at the pulse width, set the subtrim to 0 and set the PPM center to produce the same pulse width as before but also check the end points and adjust. Effectively I manually ensure the servo arm is perpendicular to the control surface at the trimmed position. This process can surely be automated.