opentx/opentx #2080: Enhancement: Set "Backup Folder" as part of the Radio Profile instead of Application Settings

CoyotteDundee, 2015-02-16:
Hi All,
I wonder if the Companion's "Backup Folder" parameter could be moved from the Application Settings to the Radio Profile, allowing having a backup folder per Radio Profile.
This could allow people having multiple models to not mix backups.

Best regards,

Coyotte

LapinFou, 2015-02-16:
:+1: Very good idea !

LapinFou, 2015-04-07:
Can't wait to try the first stable version of 2.1.
:)

romoloman, 2015-04-27:
I would prefer to have both with profile backup folder overriding the default application one if set.
If this is accepted by all members I will implement it

CoyotteDundee, 2015-04-27:
Hello,
This 'll allow usual operation mode, as also a specific mode if needed by the user. 
The best of both worlds ;-)
Thanks for the proposal.

Coyotte

bsongis, 2015-04-27:
Yes of course, thanks @romoloman for this one, I will have more time for other issues!

romoloman, 2015-05-01:
Done in profilebackupfolder branch, give it a test.

romoloman, 2015-05-06:
Isn't this issue closed ?
Feedbacks are welcome

bsongis, 2015-05-07:
You just forgot to write "Fixes #2080" in your commit :)
Thanks for your this one!

closed 2015-05-07 by bsongis