opentx/opentx #2083: Helicopter Option

robblot, 2015-02-16:
When I try to create a new model(helicopter), the helicopter option is inactive. I am using version 2.0.15. I just learned about the Taranis transmitter and want to play with the companion a bit before I decide to purchase one.

jermoo, 2015-02-16:
The Heli option in the Model Wizard is still in work. Many functions in LUA are still not available. What you are experiencing is normal. 
We have manual steps to create Heli models depending on your needs.

robblot, 2015-02-16:
Thank you... where can I find this information please?

projectkk2glider, 2015-02-16:
True, helicopter part of the new model wizard is still missing, we are still looking for a brave soul to write it. You can read http://open-txu.org/ccpm-for-taranis/ and recreate configuration yourself.

projectkk2glider, 2015-02-16:
Duplicate of #906

closed 2015-02-16 by projectkk2glider